Nuand/bladeRF #849: BladeRF unusable due to "wait_for_buffer: Timed out waiting for buf_ready" errors

akoun, 2022-01-05:
Hello
I try to test BladeRF under Pothos (built from source) and I get strange errors when using the bladerf soapy sink

[22:22:54.214000] SoapySDR: bladerf_open_with_devinfo()
[22:22:54.394000] SoapySDR: bladerf_get_serial() = 1d62a870f59e65fdd231ced37787b03a
[22:22:54.401000] SoapySDR: setSampleRate(Rx, 0, 4.000000 MHz), actual = 4.000000 MHz
[22:22:54.409000] SoapySDR: setSampleRate(Tx, 0, 4.000000 MHz), actual = 4.000000 MHz
[22:22:54.417000] SoapySDR: setSampleRate(Rx, 0, 4.800000 MHz), actual = 4.800000 MHz
[22:22:54.421000] ::1: [INFO @ fpga_common/src/lms.c:2258] Clamping frequency to 237500000Hz
[22:23:04.972000] ::1: [INFO @ fpga_common/src/lms.c:2258] Clamping frequency to 237500000Hz
[22:23:05.017000] : 0
[22:23:05.248000] ::1: [ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 1 ms
[22:23:05.249000] ::1: [ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 1 ms
[22:23:05.250000] ::1: [ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 1 ms
[22:23:05.348000] ::1: [ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 1 ms
[22:23:05.349000] ::1: [ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 1 ms
[22:23:05.350000] ::1: [ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 1 ms
[22:23:05.447000] ::1: [ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 1 ms
[22:23:05.448000] ::1: [ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 1 ms
[22:23:05.449000] ::1: [ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 1 ms
[22:23:05.547000] ::1: [ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 1 ms
.....

the flowgraph is stuck and does not do anything. In gnuradio-companion the BladeRF source behaves similarly but curiously when gr-osmosdr sink everything works nicely.

If somebody can help I'd greatly appreciate because I am completely stuck with this.
best regards
aK

akoun, 2022-01-06:
Hello, 

I was able to resolve my problem of usb transfer time-outs. Here is the workaround:
I downgraded the FPGA image to:
2019-07-31 - v0.11.0
and the FX3 firmware to:
2018-12-16 - v2.3.2

and then the errors disappeared...

I am under Debian bookworm with libbladerf 0.2021.10-2 (I was planning to downgrade this as well to 0.2019.07-7 but finally it was not necessary).

I hope this can be helpful because I chased my tail around for a while with this one.

cheers,
aK

ratzrattillo, 2022-07-04:
I am experiencing the same issue.
Is there a statement from @Nuand regarding this issue?

Downgrading cannot be a solution on the long run.

ratzrattillo, 2022-07-06:
Let me show you a simple example of a transmission, running on Windows and Linux:

`bladeRF> version

  bladeRF-cli version:        1.8.0
  libbladeRF version:         2.4.1

  Firmware version:           2.4.0-git-a3d5c55f
  FPGA version:               0.14.0 (configured from SPI flash)`

Example Code in Rust:

File main.rs:

`use futuresdr::blocks::{Apply, SoapySinkBuilder};
use futuresdr::runtime::Flowgraph;
use futuresdr::anyhow::Result;
use futuresdr::blocks::audio::Oscillator;
use futuresdr::num_complex::{Complex32};
use futuresdr::runtime::buffer::slab::Slab;
use futuresdr::runtime::Runtime;

/* Added the following to ~/.cargo/config
[target.x86_64-pc-windows-msvc]
rustflags = ["-Ctarget-cpu=native"]
[target.stable-x86_64-unknown-linux-gnu]
rustflags = ["-Ctarget-cpu=native"]
 */

const SAMPLE_RATE: f64 = 3_072_000.0;
const FREQUENCY: f64 = 330e6;

fn main() -> Result<()> {

    let mut fg = Flowgraph::new();

    let src = fg.add_block(Oscillator::new(440.0, 1., SAMPLE_RATE as f32));
    let modulator = fg.add_block(Apply::new(move |v: &f32| -> Complex32 {
        Complex32::new(*v, *v)
    }));
    let soapy_snk = fg.add_block(SoapySinkBuilder::new().freq(FREQUENCY).sample_rate(SAMPLE_RATE).gain(0.0).filter("driver=bladerf").build()); //"driver=bladerf,xb200=auto"

    fg.connect_stream_with_type(src, "out", modulator, "in", Slab::with_size(4096))?;
    fg.connect_stream_with_type(modulator, "out", soapy_snk, "in", Slab::with_size(4096))?;

    let runtime = Runtime::new();
    runtime.run(fg)?;
    Ok(())
}`

File Cargo.toml:
`[package]
name = "morse_code"
version = "0.1.0"
edition = "2021"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]
futuresdr = { version = "*", features = [ "soapy" ]}`

Dependencies required to compile and reproduce on Ubuntu 20.04:
`sudo add-apt-repository ppa:nuandllc/bladerf
sudo apt update
sudo apt install bladerf clang, llvm, libsoapysdr-dev
curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh`

Build:
`cargo build --release`

When running this program, we get teh following results and transmittion of the carrier at 330MHz stops after few seconds:
`./target/release/program
FutureSDR: INFO - bladerf_open_with_devinfo()
[INFO @ host/libraries/libbladeRF/src/helpers/version.c:79] Firmware version (v2.4.0) is newer than entries in libbladeRF's compatibility table. Please update libbladeRF if problems arise.
[INFO @ host/libraries/libbladeRF/src/helpers/version.c:103] FPGA version (v0.14.0) is newer than entries in libbladeRF's compatibility table. Please update libbladeRF if problems arise.
FutureSDR: INFO - bladerf_get_serial() = 1234
FutureSDR: INFO - setSampleRate(Rx, 0, 4.000000 MHz), actual = 4.000000 MHz
FutureSDR: INFO - setSampleRate(Tx, 0, 4.000000 MHz), actual = 4.000000 MHz
FutureSDR: INFO - setSampleRate(Tx, 0, 3.072000 MHz), actual = 3.072000 MHz
[INFO @ fpga_common/src/lms.c:1019] Clamping TXVGA2 gain to 0dB
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x7f1be40cbaf0
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x7f1be40cfb00
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x7f1be40d3b10
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:305] wait_for_buffer: Timed out waiting for buf_ready after 1000 ms
[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:305] wait_for_buffer: Timed out waiting for buf_ready after 100 ms
FutureSDR: INFO - bladerf_close()`

The same behaviour can be witnessed on Windows and on a different machine too:
The only thing that changes are the dependencies (I can provide them, if required to solve this issue).

Note, that the errors did not appear before v0.12.0. So what is the issue here? 
@Nuand , @rghilduta @rtucker , @jynik ? Do you have any clue?

I am curious, if the bladeRF repository is still maintained, as i can encounter a lot of unsolved issues without comments here on Github. Is the bladeRF still supported, or do we have to search for another platform?

jbarea, 2022-10-01:
Hello all, just got across this posts searching for answers as i updated firmware and fpga in my bladerf x40 but i'm getting timeout erros like the ones described here and cannot properly calibrate the device. Sometimes doing cal lms just exits bladeRF-cli completely. Any help or ideas will be very much appreciated as i want to get the device up and running again.
Thanks in advance!!!

DariusBotha95, 2022-11-02:
Good day, I am also facing the same issue: The timeout occurs on Tx transfers with the FPGA on the latest version, and no issue on v0.12.0 or less.
Any update on this? Why continue to sell it if there isn't support for it?

DariusBotha95, 2022-11-04:
Hi @rghilduta, upon further investigation, I found that reverting commit 2a8774d solved the issue of time-outs. I can't see any adverse effects by doing this, but let me know if I might have missed something. Hopefully, that narrows it down a bit.

Regards,
Darius