Nuand/bladeRF #858: ERROR : get_fpga_version() failed: operation timed out.

Naooss, 2022-02-22:
Hello,

i added files in baldeRF-nios sources to change the gain, frequency,,samplerate... on rx  automatically when a reception occures. But when i am trying to load bitstream on the fpga i get the error : get_fpga_version() failed : operation timed out.

I don't know how to fix this.

rghilduta, 2022-03-10:
What sort of modifications were made to bladeRF-nios? Is vanilla bladeRF-cli and FPGA image able to load?

Naooss, 2022-03-10:
Hello, thanks for your answer

i added those lines to the existing code in bladerf_nios.c :

// Frequency

#define LOC_FREQUENCY		108100000				// 108,1 MHz
#define GLIDE_FREQUENCY		329600000				// 329,6 MHz

#define REF_FREQUENCY		108100000				// 108,1 MHz
#define VAR_FREQUENCY		329600000				// 329,6 MHz

// Bandwidth

#define LOC_BANDWIDTH		50000					// 50 KHz
#define GLIDE_BANDWIDTH		50000					// 50 KHz

#define REF_BANDWIDTH		50000					// 50 KHz
#define VAR_BANDWIDTH		50000					// 50 KHz

// Gain

#define LOC_GAIN			25
#define GLIDE_GAIN			25

#define REF_GAIN			25
#define VAR_GAIN			25

// Samplerate

#define LOC_SAMPLERATE		1000000					// 1 MHz
#define GLIDE_SAMPLERATE	1000000					// 1 MHz

#define REF_SAMPLERATE		1000000					// 1 MHz
#define VAR_SAMPLERATE		1000000					// 1 MHz

/* Initialize RFIC */
    rfic_command_write_immed(BLADERF_RFIC_COMMAND_INIT, BLADERF_CHANNEL_RX(0), BLADERF_RFIC_INIT_STATE_ON);
    rfic_command_write_immed(BLADERF_RFIC_COMMAND_INIT, BLADERF_CHANNEL_RX(1), BLADERF_RFIC_INIT_STATE_ON);

 	/* Set up filters */
 	
    rfic_command_write_immed(BLADERF_RFIC_COMMAND_FILTER,BLADERF_CHANNEL_RX(0),BLADERF_RFIC_RXFIR_DEC1);
    rfic_command_write_immed(BLADERF_RFIC_COMMAND_FILTER,BLADERF_CHANNEL_RX(1),BLADERF_RFIC_RXFIR_DEC1);

  // /* Start RX */
	rfic_command_write_immed(BLADERF_RFIC_COMMAND_ENABLE, BLADERF_CHANNEL_RX(0), true);
	rfic_command_write_immed(BLADERF_RFIC_COMMAND_ENABLE, BLADERF_CHANNEL_RX(1), true);

	/* Set Frequency */
    rfic_command_write_immed(BLADERF_RFIC_COMMAND_FREQUENCY, BLADERF_CHANNEL_RX(0), LOC_FREQUENCY);
	rfic_command_write_immed(BLADERF_RFIC_COMMAND_FREQUENCY, BLADERF_CHANNEL_RX(1), LOC_FREQUENCY);
	

	// /* Set Bandwidth */
	rfic_command_write_immed(BLADERF_RFIC_COMMAND_BANDWIDTH, BLADERF_CHANNEL_RX(0), LOC_BANDWIDTH);
	rfic_command_write_immed(BLADERF_RFIC_COMMAND_BANDWIDTH, BLADERF_CHANNEL_RX(1), GLIDE_BANDWIDTH);

	// /* Set Gain */
	rfic_command_write_immed(BLADERF_RFIC_COMMAND_GAINMODE, BLADERF_CHANNEL_RX(0), BLADERF_GAIN_MGC);
	rfic_command_write_immed(BLADERF_RFIC_COMMAND_GAINMODE, BLADERF_CHANNEL_RX(1), BLADERF_GAIN_MGC);

	rfic_command_write_immed(BLADERF_RFIC_COMMAND_GAIN, BLADERF_CHANNEL_RX(0), LOC_GAIN);
	rfic_command_write_immed(BLADERF_RFIC_COMMAND_GAIN,BLADERF_CHANNEL_RX(1), GLIDE_GAIN);

	// /* Set Sample Rate */
	rfic_command_write_immed(BLADERF_RFIC_COMMAND_SAMPLERATE, BLADERF_CHANNEL_RX(0), LOC_SAMPLERATE);
	rfic_command_write_immed(BLADERF_RFIC_COMMAND_SAMPLERATE, BLADERF_CHANNEL_RX(1), GLIDE_SAMPLERATE);

I am now able to load the bitstream but when i do a reception it does not change frequency, gain ect .... it is still default values.

And yes vanilla bladeRF-cli and FPGA image is able to load.