Nuand/bladeRF #863: AGC in BladeRF XA9

SameerIrfan, 2022-03-30:
Hi,
I'm using bladeRF xA9 to transmit and receive using GNU Radio 3.8 but when I change the gain in the Osmocom blocks there is no change in the plots. I think this is because of the automatic gain control in the BladeRF. can any one help how to turn off AGC in gnu radio.
P.S. I'm a newbie hence doesn't now much about SDRs.

robrechtrosseel, 2022-03-30:
In the osmocom block, I use this as device argument: bladerf=0,biastee=0,agc_mode=manual,agc=0
And this does the trick for me. To change the gain, I change RF Gain (dB) and don't touch the IF and BB gain.

SameerIrfan, 2022-03-30:
@robrechtrosseel i have tried to use this as device argument but still I am getting the same results as before.

robrechtrosseel, 2022-03-30:
[TransmitCW.grc.zip](https://github.com/Nuand/bladeRF/files/8379837/TransmitCW.grc.zip)
Can you try with this simple example?

SameerIrfan, 2022-04-01:
@robrechtrosseel in the example you have given above when I change the Tx gain nothing happens. Also, I see the same graph when the receiver has a dummy load attached which I see when the transmitter is connected to it.
For the transmitter part I have attached the frequency analyzer to the transmitter port but nothing happens when I change the Tx gain. 
Also, when Tx is enabled there are peaks at 3GHz, 5GHz etc. like the ones I was getting in my flowgragh (like i have mentioned before in the original comment), could you also tell me why am i seeing those peaks?

robrechtrosseel, 2022-04-01:
@SameerIrfan I'm sorry that it's not working with you machine/bladeRF. I only have the A4 and A5 and both work fine. I can't help you with the different peaks.