Nuand/bladeRF #865: BladeRF xA9

user11422, 2022-04-20:
hello
I've recently got my hands on the BladeRF and being new I don't know anything about it. I've noticed 4 lights on it that blink with some pattern. could anyone please tell me what those lights indicate and what lights should be lit when I am transmitting or receiving a signal? Also, can anyone send me or direct me to where I can find the datasheet for the xA9.
Regards

ax-initrd, 2022-04-29:
It depends on the FPGA image you use. Take a look on the VHD file of the used image and search for "led(". For the default image led2 is used for "tx_underflow" and led3 for "rx_overflow" respectively.

rthomp10, 2023-01-17:
The LEDs represent certain states within the bladeRF. For instance, the bladerf-hosted image is set to the following.

LED 1: Engagement of the FX3 signifying active USB transfers
LED 2: TX underflow
LED 3: RX overflow

Seen in [/platforms/bladerf-micro/vhdl/bladerf-hosted.vhd](https://github.com/Nuand/bladeRF/blob/master/hdl/fpga/platforms/bladerf-micro/vhdl/bladerf-hosted.vhd#L480-L483)

rthomp10, 2023-01-17:
The datasheet can be found in the [bladeRF 2.0 micro xA9 product page](https://www.nuand.com/product/bladeRF-xA9/).

closed 2023-01-17 by rthomp10