Nuand/bladeRF #867: Modulate and Demodulate a signal in GRC

SameerIrfan, 2022-04-21:
Hi,
I'm still learning to use BladeRF xA9 with GNU Radio Companion and wanted to modulate a signal and send it through a RF Cable and receive it on the receiver side and demodulate to retrieve the original message. Can anyone help me with this?
Thanks in Advance

raul8883, 2022-06-30:
Hi, here are some analog modulations. Because i can't upload the .grc file i will upload images
The way in which I have done the modulations has been with its low pass equivalent (in phase components, cuadrature components), because the modulations require high frequencies, and the sampling frequency must be at least double, which is not feasible for simple computers, the SDR will take care of going up in frequency multiplying these signals (in phase components and in quadrature) by two tones 90º out of phase with each other, the frequency of these tones will be the carrier frequency.

                                                              AM
AM_TX
[image]
AM_RX
[image]
                                                             LSB
LSB_TX
[image]
LSB_RX
[image]
                                                            USB
USB_TX
[image]
USB_RX
[image]

I cant upload the message (.wav file) it has fs = 15000Hz
Be careful not to saturate with the gain of the SDR, values ​​between [-1,1]

SameerIrfan, 2022-07-01:
Hi, thanks for your reply. the pictures you have sent are great but I was hoping for help in Frequency Modulation is you could do it  please help me in that. 
What I actually want to do is first modulate the signal using frequency modulation and then send it via an antenna and receive it on another BladeRF and demodulate to obtain the original message/ signal. 
Please help me in this 
Regards

raul8883, 2022-07-01:
GNU radio already provides the necessary blocks for FM modulation and demodulation, in my case, I have used the WBFM transmit block, the output of this block already offers the components in phase and quadrature corresponding to FM
in-phase component yF(t) = Ac * cos(InstantPhase);
cuadrature component yC(t) = Ac * sen(InstantPhase);
Where; 
InstantPhase (t) = 2 * pi * frecuencyDeviation * integral( x (t));
TX_FM
[image]
RX_FM
[image]

The message used has a sampling frequency of 22050 Hz, therefore the maximum frequency of my message is around 11000Hz  = (fmax = fs/2)
As for the value of the maximum deviation (fd) used in the WBFM transmit block, it has been 55KHz, because I have taken a value of Beta = frequency deviation / frequency of the modulator = 5 = fd / 11000 Hz.
So make sure that the receive equivalent lowpass filter has a high enough cutoff frequency so as not to remove relevant information, since FM modulations (angular modulations) have a higher bandwidth than linear modulations.

raul8883, 2022-07-01:
By the way are you going to use two BladeRF (one to transmit and one to receive or are you going to use only one as transmitter and receiver?

SameerIrfan, 2022-07-01:
For the initial testing I am using 2 BladeRFs but I intend to use only 1 as a MIMO in the future

raul8883, 2022-07-01:
I say this because I have commented on a problem about using 2 bladerf it will probably happen to you too.
https://github.com/Nuand/bladeRF/issues/875

SameerIrfan, 2022-07-01:
In the screenshots you have posted above you have used a WAV file to actually hear the message. Could you tell me if I use a single Frequency instead of a WAV file in the above flowgraph will it still work or will i need to make any changes in it? Also, if i can some how make the signal reflect from a metal sheet and vibrate the sheet can i find its doppler shift?

raul8883, 2022-07-01:
Make sure that the tone you use as a source has a frequency of 11000Hz and fs = 22050, with this you will not have to modify any other parameters.
As for the doppler shift, I haven't studied anything related to that, but I suppose that if you do it correctly based on the theory you will have no problem finding said doppler shift.

SameerIrfan, 2022-07-01:
> I say this because I have commented on a problem about using 2 bladerf it will probably happen to you too. #875

yes I have also seen this but I couldn't figure out what was the problem actually. I thought there was some problem with my flowgraph.

raul8883, 2022-07-01:
I guess for your project this should not be a big problem.

SameerIrfan, 2022-07-02:
@raul8883 i have tried to make the FM modulation flowgraph like the one you have sent above but i am not getting the message back. All i hear are pulses of noise. Could you please look at my flowgraph and check of there are any errors. 
Also, for the sample rate 22.05k it is giving the error:
"RuntimeError: bladerf_common::set_sample_rate: Failed to set sample rate: Provided parameter was out of the allowable range (-2)"
so I changed the sample rate to 100k just so i can make the flowgraph run and hear anything

raul8883, 2022-07-02:
You should sample the message at least twice the maximum frequency of the signal, after this you should interpolate your signal, that is, increase the sample frequency and adapt it to your SDR, to know what is the allowed range of your SDR sample , run the bladerf cli and type 'print' there are the ranges within which your SDR is capable of working.

raul8883, 2022-07-02:
Then in reception you must decimate your received signal, that is, reduce your sampling frequency again to recover your original signal and be able to adapt it to your sound card, sound cards can normally work with sampling frequencies of 44100Hz.
the human ear in general can capture sounds between 20 Hz and 20000Hz hence the sampling frequency normally used in sound cards or audio players is fs>=fmax*2 =~ 44100 Hz

SameerIrfan, 2022-07-02:
@raul8883 I'm still not getting the message back. Could you take a quick look on my flowgraphs.
P.s. I have also added the message wav file which I am trying to send
[FM transmitter and receiver.zip](https://github.com/Nuand/bladeRF/files/9033774/FM.transmitter.and.receiver.zip)
.

raul8883, 2022-07-02:
Can you send a picture?

SameerIrfan, 2022-07-03:
@raul8883 here are the pictures
[image]
[image]

SameerIrfan, 2022-07-03:
@raul8883 Also, is there any way I can contact you in private? I want to discuss something about the BladeRF and GNURadio.

SameerIrfan, 2022-07-05:
@raul8883 if your free and know anything about the problem i sent before could you please help me as i am really stuck at it
i have also posted pictures of my flowgraph could you please review it?

raul8883, 2022-07-06:
Hello, I am currently quite saturated with a project, but maybe I can help, the first error I see is a receiver-transmitter scheme, I mean it would be better to draw a transmitter-receiver scheme

raul8883, 2022-07-06:
Another error is the gains of the transmitting and receiving antennas, it makes no sense to indicate negative values, what i would do is look at the specs of your bladerf and check the range of gains allowed, finally all the sample rates are wrong so you are not getting your original message back.

The way I learned and I suggest you do something similar is, first try to inform yourself about how an SDR works and what ranges it can work on, your specific SDR, after this, inform yourself well about the operation of the GNU blocks radius, and finally learn about the nyquist criterion (fs >= 2*fmax)