Nuand/bladeRF #880: bladeRF-fsk Update?

AgileEngineeringLLC, 2022-08-14:
I'm trying to run the bladeRF-fsk, and I'm getting the following error:

Failed to set TX VGA1 gain: Operation not supported

Looks like a cool demo. I'm guessing bladeRF-fsk may not be updated for BladeRF 2 devices? If so, any plans to update? Any advice on what I'd need to change to get it working on my own?

Thanks!

rghilduta, 2022-08-24:
Please take a look at branch [dev-fsk](https://github.com/Nuand/bladeRF/tree/dev-fsk) for a fix that extends support to unified gains (which is what the bladeRF 2.0 micro uses).

Maeven-Black, 2022-09-03:
Verified on linux. You will need to build from source for now:
1) Uninstall the make and delete the build directory
2) Change the git branch to dev-ask
3) Reinstall using the cmake instructions

From a clean install:
```
 git clone https://github.com/Nuand/bladeRF.git ./bladeRF
 cd ./bladeRF
 cd host
 git switch dev-fsk
 mkdir build
 cd build
 cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr/local -DINSTALL_UDEV_RULES=ON ../
 make
 sudo make install
 sudo ldconfig
```

AgileEngineeringLLC, 2022-09-03:
Hi,

I was able to get it built per the instructions you sent, but I tried both demos (with default gains) and I’m getting:

Sent: 0% ERROR: Transmission failed (no connection)

What was your exact hardware setup, and the commands you used to get it to work?

Thx!!

P.S. My setup is cabled, with
 RX side: bladeRF 2.0 with DCblock and 20 dB attenuator
 TX side: bladeRF 2.0 with DCblock and 20 dB attenuator

 TX1 and RX1 of RX bladeRF are connected to RX1 and TX1 of TX bladeRF respectively

Don

raf329, 2023-03-13:
> Hi,
>
> I was able to get it built per the instructions you sent, but I tried both demos (with default gains) and I’m getting:
>
> Sent: 0% ERROR: Transmission failed (no connection)
>
> What was your exact [...]

Did you solve this Error? 
I have the same problem. 
My settings:
bladerf-fsk -d*:xxxx1 -r904M -t 924M --tx-vga2 5
bladerf-fsk -d*:xxxx -r924M -t 904M --tx-vga2 5

AgileEngineeringLLC, 2023-03-13:
No, i wasn't.

raf329, 2023-03-13:
> No, i wasn't.

Hm, could you suggest me some another tools for realise chatting between 2 SDR boards?

ifrasch17, 2025-06-14:
Hi there! bladeRF-fsk has now been updated to fully support the bladeRF 2.0 devices, among other things. There will be a blog post coming soon with more info. Please give it a try with the latest bladeRF codebase and be sure to update to the latest FPGA image as well.

Let us know if you have any further issues.

closed 2025-06-14 by ifrasch17