Nuand/bladeRF #882: XB200 unnecessary error thrown

ratzrattillo, 2022-10-15:
I noticed that when using the XB-200 expansion board and setting the frequency always the following error will be thrown:
```
[ERROR @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:2398]
Consider supplying the quick_tune parameter to bladerf_schedule_retune() when the XB-200 is enabled.
```

This is due to the function [bladerf1_set_frequency](https://github.com/Nuand/bladeRF/blob/master/host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c#L2069) always calling [schedule_retune](https://github.com/Nuand/bladeRF/blob/master/host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c#L2069) with NULL as the quicktune parameter, when the tuning mode is BLADERF_TUNING_MODE_FPGA:
https://github.com/Nuand/bladeRF/blob/master/host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c#L2069

The function [bladerf1_schedule_retune](https://github.com/Nuand/bladeRF/blob/master/host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c#L2372) always checks, if the quicktune parameter is NULL, and throws the error above, if the XB-200 is attached.

Thus, the error can never be avoided, because as soon as calling set_frequency with an XB-200 attached, the error will be thrown.

I propose to get the quicktune parameters and supplying them before calling [schedule_retune](https://github.com/Nuand/bladeRF/blob/master/host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c#L2069), when the XB-200 is attached.

This could be a solution replacing line [2069](https://github.com/Nuand/bladeRF/blob/master/host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c#L2069) in bladerf1.c:

```C
if (attached == BLADERF_XB_200) {
    bladerf_quick_tune qt;
    status = bladerf_get_quick_tune(dev, ch, &qt);
    if (status != 0) {
        return status;
    }
    status = dev->board->schedule_retune(dev, ch, BLADERF_RETUNE_NOW, 0, &qt);
}
else {
    status = dev->board->schedule_retune(dev, ch, BLADERF_RETUNE_NOW, frequency, NULL);
}
```
What do you think? Looking forward to your feedback!

ratzrattillo, 2022-10-15:
Added a pull request: https://github.com/Nuand/bladeRF/pull/883

rthomp10, 2023-01-17:
I believe changing the log message to INFO serving as a reminder is the right way to go. The issue will be fixed in the next release.

commit fe160fa5 (2023-03-31): libbladeRF: changed xb200 quick tune param recommendation

XB-200 quick tune parameter recommendation now considered log info rather 
than an error.

Fix #882
commit 7825ae5c (2023-03-31): libbladeRF: changed xb200 quick tune param recommendation

XB-200 quick tune parameter recommendation now considered log info rather 
than an error.

Fix #882
closed 2023-03-31 by rthomp10