Nuand/bladeRF #884: Implement BLADERF_FORMAT_SC12_Q11

Ka-zam, 2022-10-17:
Hi,

would it be possible/desirable to implement an interleaved SC12_Q11 format?

This way the bitrate over USB3 could be reduced at the cost of host CPU power.
My aim is to stream 2 channels at 61.44 Msps and I find that it is not possible with the standard SC16_Q11 sign extended format.

rghilduta, 2023-02-28:
We have added 8bit mode. The problem with SC12_Q11 is that it does not evenly fit into USB packets. This may be added at a later point but for the time being the library supports 12bit mode sign extended to 16bits, or 12bit samples truncated to 8bits (BLADERF_FORMAT_SC8_Q7).
Please see: https://nuand.com/libbladeRF-doc/v2.5.0/group___s_t_r_e_a_m_i_n_g___f_o_r_m_a_t.html#gga4c61587834fd4de51a8e2d34e14a73b2a43f18d5f42d2cb3c802c3669fd7d529a

manoskav, 2023-03-22:
Hello,
does BLADERF_FORMAT_SC8_Q7_META really work? I am trying to use it, but I get overruns in all bladerf_sync_rx calls. I am running the example [https://github.com/Nuand/bladeRF/blob/master/host/libraries/libbladeRF/doc/examples/sync_rx_meta.c](https://github.com/Nuand/bladeRF/blob/master/host/libraries/libbladeRF/doc/examples/sync_rx_meta.c) where I only change the format in bladerf_sync_config call. Also, the scheduled sync_rx part of the example seems to get stuck.

I am using bladeRF 2.0 micro xA4 on a raspberry pi 4.
Everything is on latest version as seen below from bladeRF-cli:
```
bladeRF> version

  bladeRF-cli version:        1.9.0-git-12699a5c
  libbladeRF version:         2.5.0-git-12699a5c

  Firmware version:           2.4.0-git-a3d5c55f
  FPGA version:               0.15.0 (configured from SPI flash)
```

I get the following output when executing the test:

```
Opening and initializing device...

RX frequency: 1575420000 Hz
RX samplerate: 1000000 sps
RX bandwidth: 1000000 Hz
RX gain: 60
TX frequency: 1575420000 Hz
TX samplerate: 1000000 sps
TX bandwidth: 1000000
TX gain: 60

Running RX meta "now" example...
Overrun detected. 1016 valid samples were read.
Overrun detected. 1016 valid samples were read.
Overrun detected. 1016 valid samples were read.
Overrun detected. 1016 valid samples were read.
Overrun detected. 1016 valid samples were read.
Overrun detected. 1016 valid samples were read.
Overrun detected. 1016 valid samples were read.
Overrun detected. 1016 valid samples were read.
Overrun detected. 1016 valid samples were read.
Overrun detected. 1016 valid samples were read.
Overrun detected. 1016 valid samples were read.
Overrun detected. 1016 valid samples were read.
Overrun detected. 1016 valid samples were read.
Overrun detected. 1016 valid samples were read.
Overrun detected. 1016 valid samples were read.

Running RX meta "scheduled" example...
Current RX timestamp: 0x000000000000448e
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x55a77db230
Scheduled RX failed: Operation timed out
```

BLADERF_FORMAT_SC16_Q11_META works as expected.

KarlL2, 2023-04-12:
This has already been discussed several years ago in https://github.com/Nuand/bladeRF/issues/685 , with more info on 12-bit alignment. SC12 should indeed allow dual channel at 61.44 Msps.

cwozny, 2023-09-10:
> Hello,
>
> does BLADERF_FORMAT_SC8_Q7_META really work? I am trying to use it, but I get overruns in all bladerf_sync_rx calls. I am running the example [...]

Did you enable to 8-bit overclock mode via bladerf_enable_feature(BLADERF_FEATURE_OVERSAMPLE)?

KarlL2, 2026-08-27:
@rthomp10 From what I see, this is only possible without metadata right, meaning we cannot have timestamp info, right?