Nuand/bladeRF #893: hdl build_bladerf.sh script failing for bladeRF-micro A9 hosted

t1ch, 2022-12-18:
I am encountering the following error when running `./build_bladerf.sh -b bladeRF-micro -s A9 -r hosted`

`make: *** No rule to make target '../../../../../../thirdparty/analogdevicesinc/no-OS/ad9361/sw/*', needed by '../../../../../quartus/work/bladerf-micro-A9-hosted/libad936x/Makefile'.  Stop.`

The script works when called as follows  `./build_bladerf.sh -b bladeRF -s 115 -r hosted` 
Is there something I am overlooking?

Nuand, 2023-01-05:
Please try please to build the host/ libraries and tools first, https://github.com/Nuand/bladeRF/wiki/Getting-Started%3A-Linux#building-bladerf-libraries-and-tools-from-source . In the process, cmake should generate the libad936x needed to compile the C source coded needed for the FPGA image.

closed 2023-01-05 by Nuand

Skeletoskull, 2024-02-15:
did your issue got resolved?
because I tried using the link and didnt work for me

t1ch, 2024-02-17:
@Skeletoskull Yes it got resolved. You have to build the bladeRF tools first

Skeletoskull, 2024-02-17:
> @Skeletoskull Yes it got resolved. You have to build the bladeRF tools first

Do you have a youtube tutorial or any link because I am totally new to this and have no idea what you just said. 
The link above is also broken.

Semnodime, 2024-05-29:
The `thirdparty/analogdevicesinc/no-OS` is a git submodule refering to https://github.com/analogdevicesinc/no-OS/tree/0bba46e6f6f75785a65d425ece37d0a04daf6157.
To resolve the issue, clone https://github.com/Nuand/bladeRF/ with its external submodules via `git clone --recurse-submodules https://github.com/Nuand/bladeRF`.

Skeletoskull, 2024-05-29:
I did it, thanks