Nuand/bladeRF #894: Cann't update the firmware

YY-chen555, 2022-12-30:
When I use windows(my physical machine) to update the firmware.It doesn't work.

`F:\bladeRF\x64>bladeRF-cli.exe -f F:\bladeRF\fx3_images\bladeRF_fw_v1.9.1.img -v verbose
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:642] Using libusb version: 1.0.22.11312
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:519] Found a bladeRF (idx=4)
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/libusb.c:191] Bus 001 Device 014: Nuand bladeRF, serial 04d36bb5eb36580fe2fba49f451075b2
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:409] Mutex name: Global\bladeRF-04d36bb5eb36580fe2fba49f451075b2
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 0
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:851] Read Firmware version: 2.4.0-git-a3d5c55f
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:860] Capability mask before FPGA load: 0x0000007700000000
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:385] Found SPI flash manufacturer: MACRONIX.
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:388] Found SPI flash device: MX25U3235E (32 Mbit).
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:445] SPI flash total size = 32 Mbit
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:446] SPI flash page size = 256 bytes
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:447] SPI flash erase block size = 65536 bytes
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:448] SPI flash number of pages = 16384
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:449] SPI flash number of erase blocks = 64 pages
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 2
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/usb.c:131] Reporting no FPGA present - BLADERF_FORCE_NO_FPGA_PRESENT is set.
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 3
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 2
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/usb.c:131] Reporting no FPGA present - BLADERF_FORCE_NO_FPGA_PRESENT is set.
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 3
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:990] Skipping FPGA configuration and initialization - BLADERF_FORCE_NO_FPGA_PRESENT is set.
Flashing firmware...
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 2
[INFO @ host/libraries/libbladeRF/src/backend/usb/usb.c:505] Erasing 3 blocks starting at block 0
[INFO @ host/libraries/libbladeRF/src/backend/usb/usb.c:510] Erasing block 2 (100%)...
[INFO @ host/libraries/libbladeRF/src/backend/usb/usb.c:520] Done erasing 3 blocks
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 1
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 2
[INFO @ host/libraries/libbladeRF/src/backend/usb/usb.c:732] Writing 474 pages starting at page 0
[INFO @ host/libraries/libbladeRF/src/backend/usb/usb.c:738] Writing page 473 (100%)...
[INFO @ host/libraries/libbladeRF/src/backend/usb/usb.c:747] Done writing 474 pages
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 1
[INFO @ host/libraries/libbladeRF/src/driver/spi_flash.c:114] Verifying 474 pages, starting at page 0
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 2
[INFO @ host/libraries/libbladeRF/src/backend/usb/usb.c:620] Reading 474 pages starting at page 0
[INFO @ host/libraries/libbladeRF/src/backend/usb/usb.c:625] Reading page 473 (100%)...
[INFO @ host/libraries/libbladeRF/src/backend/usb/usb.c:636] Done reading 474 pages
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 1
Successfully wrote firmware to flash!
NOTE: A power cycle is required to load the new firmware.
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:642] Using libusb version: 1.0.22.11312
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:519] Found a bladeRF (idx=4)
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/libusb.c:191] Bus 001 Device 014: Nuand bladeRF, serial 04d36bb5eb36580fe2fba49f451075b2
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:409] Mutex name: Global\bladeRF-04d36bb5eb36580fe2fba49f451075b2
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 0
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:851] Read Firmware version: 2.4.0-git-a3d5c55f
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:860] Capability mask before FPGA load: 0x0000007700000000
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:385] Found SPI flash manufacturer: MACRONIX.
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:388] Found SPI flash device: MX25U3235E (32 Mbit).
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:445] SPI flash total size = 32 Mbit
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:446] SPI flash page size = 256 bytes
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:447] SPI flash erase block size = 65536 bytes
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:448] SPI flash number of pages = 16384
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:449] SPI flash number of erase blocks = 64 pages
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 2
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 1
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 2
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 1
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 1
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:556] Read FPGA version: 0.12.0
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:561] Capability mask after FPGA load: 0x0000007700001fff
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:429] nios_config_read: Read 0x00180000
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/bladerf1.c:640] Default GPIO value found - initializing device
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:440] nios_config_write: Wrote 0x000000d7
[VERBOSE @ host/libraries/libbladeRF/src/driver/si5338.c:528] Setting integer sample rate: 1000000
[VERBOSE @ host/libraries/libbladeRF/src/driver/si5338.c:377] Found r value of: 4
[VERBOSE @ host/libraries/libbladeRF/src/driver/si5338.c:387] MSx a + b/c: 316 + 4/5
[VERBOSE @ host/libraries/libbladeRF/src/driver/si5338.c:416] MSx a + b/c: 316 + 4/5
[VERBOSE @ host/libraries/libbladeRF/src/driver/si5338.c:205] MSx P1: 0x00009c66 (40038) P2: 0x00000002 (2) P3: 0x00000005 (5)
[VERBOSE @ host/libraries/libbladeRF/src/driver/si5338.c:341] Calculated multisynth frequency: 1000000 + 0/1
[VERBOSE @ host/libraries/libbladeRF/src/driver/si5338.c:228] Writing MS2
[ERROR @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:69] Failed to send NIOS II request: File or device I/O failure
[DEBUG @ host/libraries/libbladeRF/src/driver/si5338.c:69] Could not read from si5338 (-5): File or device I/O failure
[VERBOSE @ host/libraries/libbladeRF/src/driver/si5338.c:545] Set actual integer sample rate: 1000000
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/libusb.c:950] lusb_control_transfer failed: status = -1
[ERROR @ host/libraries/libbladeRF/src/backend/usb/usb.c:239] Failed to switch to NULL interface: File or device I/O failure
Failed to open device (first available): File or device I/O failure`

And then,I try to know whether  the version is too old.
`F:\bladeRF\x64>bladeRF-cli.exe -e version

No bladeRF device(s) available.

If one is attached, ensure it is not in use by another program
and that the current user has permission to access it.

  bladeRF-cli version:        1.8.0
  libbladeRF version:         2.4.0

  Device version information unavailable: No device attached.`

Whenever updating the firmware fails, my physical chances won't start the bladeRF again. Windows prompt: cannot find the corresponding driver.
I don't know how to fix it. I  try to use ubuntu20.04(VM) and it also doesn't work.
In Ubuntu, the message as the following code,
`please update libbladerf if problems arise`
And I try to run this
`apt install libbladerf-dev`
I think I have installed the latest library.

Nuand, 2023-01-05:
Are you using the official Nuand Windows installer or are you building from source?
It appears the Ubuntu installation is running a VM. Please see the virtual machine FAQ on the Nuand site: https://www.nuand.com/frequently-asked-questions/#Does_the_bladeRF_run_in_a_Virtual_Machine .
Since the errors appear similar it maybe an XHCI issue, would you potentially have another Linux computer to try this on?

YY-chen555, 2023-01-06:
> 

Thanks for your reply, I checked the site and made sure XHCI was enabled. when I switched to another VM everything worked again, I wasn't sure if there was something wrong with the previous ubuntu.
I'm using the official windows software, unfortunately bladeRF doesn't work on my physical machine (windows)

Meanwhile, I would like to ask another question. I want to use OpenBts+Blade RF to complete my pseudo-base station experiment, but the tutorials on the official website seem to be too old. Does Blade RF support OpenBts now?

I use this [tutorials](https://github.com/Nuand/bladeRF/wiki/Minimalistic-build-and-run-test-for-OpenBTS-5)
When I do this step,something is wrong.
[image]

`./build.sh && \
> dpkg -i *.deb;
rm -rf coredumper-1.2.1
tar zxf coredumper-1.2.1.tar.gz
cd coredumper-1.2.1
mv packages/deb debian
chmod 644 debian/control
patch -p0
patching file src/elfcore.c
patching file src/linuxthreads.c
patching file src/linux_syscall_support_unittest.cc
dpkg-buildpackage -us -uc
dpkg-buildpackage: info: source package libcoredumper
dpkg-buildpackage: info: source version 1.2.1-1
dpkg-buildpackage: info: source distribution unstable
dpkg-buildpackage: info: source changed by Google Inc. <opensource@google.com>
dpkg-buildpackage: info: host architecture amd64
 dpkg-source --before-build .
 debian/rules clean                                                          
dh_testdir                                                                   
dh_testroot
rm -f build-stamp 
# Add here commands to clean up after the build process.
/usr/bin/make distclean
make[1]: Entering directory '/home/a/Wireless/libcoredumper/coredumper-1.2.1'
make[1]: *** No rule to make target 'distclean'.  Stop.
make[1]: Leaving directory '/home/a/Wireless/libcoredumper/coredumper-1.2.1'
make: [debian/rules:59: clean] Error 2 (ignored)
cp -f /usr/share/misc/config.sub config.sub
cp -f /usr/share/misc/config.guess config.guess
dh_clean 
dh_clean: error: Compatibility levels before 7 are no longer supported (level 4 requested)
make: *** [debian/rules:62: clean] Error 255
dpkg-buildpackage: error: debian/rules clean subprocess returned exit status 2
ERROR: command failed!
`
I don't know how to fix it. I checked the information, and someone said to change the value of the compat package from 4 to 9. But for me, it did not work.

I'm not sure it's well to ask this problem in this issue.
Thank you for your help and look forward to your reply soon!

rghilduta, 2023-02-28:
Please consider using YateBTS instead, instructions are provided here: https://nuand.com/forums/viewtopic.php?t=5094 . Additional instructions and tarballs can be downloaded from https://www.nuand.com/support/

closed 2023-02-28 by rghilduta