Nuand/bladeRF #895: cant install fpga

isayeter, 2023-01-01:
`bladeRF-cli -v verbose -i -l  /home/parallels/Downloads/hostedxA9-latest.rbf
`
```
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:642] Using libusb version: 1.0.25.11696
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:519] Found a bladeRF (idx=6)
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/libusb.c:189] Bus 001 Device 015: Nuand bladeRF 2.0, serial 729d7f533a99429cb86924309bbd34c4
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:581] USB port reset succeeded for bladeRF 729d7f533a99429cb86924309bbd34c4
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 0
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:293] Read Firmware version: 2.4.0-git-a3d5c55f
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:299] Capability mask before FPGA load: 0x0000007f00000000
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:401] Found SPI flash manufacturer: WINBOND.
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:409] Found SPI flash device: W25Q128JV (128 Mbit).
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:447] SPI flash total size = 128 Mbit
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:448] SPI flash page size = 256 bytes
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:449] SPI flash erase block size = 65536 bytes
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:450] SPI flash number of pages = 65536
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:451] SPI flash number of erase blocks = 256 pages
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 2
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/usb.c:130] Reporting no FPGA present - BLADERF_FORCE_NO_FPGA_PRESENT is set.
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 3
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:407] Skipping FPGA configuration and initialization - BLADERF_FORCE_NO_FPGA_PRESENT is set.
Loading fpga...
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 3
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 1
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:125] Read FPGA version: 0.14.0
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:131] Capability mask after FPGA load: 0x0000007f00001abc
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:586] nios_ina219_write: Wrote 0x8000 to addr 0x00
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:573] nios_ina219_read: Read 0x399f from addr 0x00
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:586] nios_ina219_write: Wrote 0x019f to addr 0x00
[DEBUG @ host/libraries/libbladeRF/src/driver/ina219.c:70] Configuration register: 0x019f
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:586] nios_ina219_write: Wrote 0xa000 to addr 0x05
[DEBUG @ host/libraries/libbladeRF/src/driver/ina219.c:82] Calibration register: 0xa000
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/common.c:270] Default tuning mode: Host
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2287] bladerf2_set_tuning_mode: New tuning mode: Host
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2337] bladerf2_set_tuning_mode: Initializing Host RFIC control
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/rfic_host.c:85] _rfic_host_initialize: initializating
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:429] nios_config_read: Read 0x00000001
[ERROR @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:77] Failed to receive NIOS II response: An unexpected error occurred
Read Error -5
[ERROR @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:68] Failed to send NIOS II request: An unexpected error occurred
Write Error -5
ad9361_init : AD936x initialization error
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/rfic_host.c:99] _rfic_host_initialize: ad9361_init(&phy, (AD9361_InitParam *)board_data->rfic_init_params, dev) failed: An unexpected error occurred
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:187] _bladerf2_initialize: dev->board->set_tuning_mode(dev, default_tuning_mode(dev)) failed: An unexpected error occurred
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2197] bladerf2_load_fpga: _bladerf2_initialize(dev) failed: An unexpected error occurred
Error: failed to load FPGA: An unexpected error occurred
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/libusb.c:950] lusb_control_transfer failed: status = -9
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/libusb.c:950] lusb_control_transfer failed: status = -9
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/libusb.c:950] lusb_control_transfer failed: status = -9
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1175] Clearing RX1 retune queue.
[ERROR @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:68] Failed to send NIOS II request: An unexpected error occurred
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/libusb.c:950] lusb_control_transfer failed: status = -9
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1175] Clearing TX1 retune queue.
[ERROR @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:68] Failed to send NIOS II request: An unexpected error occurred
[ERROR @ host/libraries/libbladeRF/src/backend/usb/usb.c:238] Failed to switch to NULL interface: An unexpected error occurred

```

isayeter, 2023-01-02:
and now, I cant access bladeRF even,

`bladeRF-cli -i`
```

[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:122] _bladerf2_initialize: dev->backend->get_fpga_version(dev, &board_data->fpga_version) failed: An unexpected error occurred
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:468] bladerf2_open: _bladerf2_initialize(dev) failed: An unexpected error occurred
[ERROR @ host/libraries/libbladeRF/src/backend/usb/usb.c:238] Failed to switch to NULL interface: An unexpected error occurred
Failed to open device (first available): An unexpected error occurred

```

rghilduta, 2023-02-28:
This appears to be a USB communication issue, are you running in a VM? If so, can you please try on a native machine?

closed 2023-07-07 by rghilduta