Nuand/bladeRF #896: Synchronized Tx/Rx of multiple bladeRFs xA4

maurokenny, 2023-01-06:
I've read the [Synchronized TRX](https://github.com/Nuand/bladeRF/tree/master/host/examples/bladeRF-cli/sync_trx) README and it seems to be possible to synchronize two or more bladerfs.

Unfortunately for me, the last commit [d3bc3dc](https://github.com/Nuand/bladeRF/commit/d3bc3dc27ff0803b4343238edc5da17bee2c6c70) has already 6 years old and it seems to use the previous version bladerf x40.

- Can you please confirm that it is still possible to sync four bladeRF 2.0 micro xA4?
- Which pins are used to connect the master and the slave boards? Pins are different between versions.
- Do you have a tutorial for that?

Thanks,

paulhprobert, 2023-01-11:
correction - that MINI_EXP_1 pin has to stay low until your tx and rx processes are done, its not just an edge trigger.

rthomp10, 2023-01-13:
The libbladeRF docs offer a description and example of the bladeRF 2.0 micro's trigger functionality [here](https://nuand.com/bladeRF-doc/libbladeRF/v2.2.1/group___f_n___t_r_i_g.html). Use the MINI_EXP_1 pin on J51-1. The Synchronized TRX README has also been updated at commit d9c115cc783350a4e1f729c3af58991248f2e77b.

closed 2023-01-13 by rthomp10