Nuand/bladeRF #899: AD936x initialization error, no chance to load fpga

lirrel, 2023-01-24:
No chance to load fpga on BladeRF 2.0 A4, firmware version installed: 2.4.0

`bladeRF-cli -l hostedxA4.rbf `

Output:
```
Loading fpga...
ad9361_init : Unsupported PRODUCT_ID 0x0
ad9361_init : AD936x initialization error
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/rfic_host.c:99] _rfic_host_initialize: ad9361_init(&phy, (AD9361_InitParam *)board_data->rfic_init_params, dev) failed: An unexpected error occurred
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:187] _bladerf2_initialize: dev->board->set_tuning_mode(dev, default_tuning_mode(dev)) failed: An unexpected error occurred
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:2203] bladerf2_load_fpga: _bladerf2_initialize(dev) failed: An unexpected error occurred
Error: failed to load FPGA: An unexpected error occurred
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1945] bladerf2_trigger_init: Board state insufficient for operation (current "FPGA Loaded", requires "Initialized").
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1945] bladerf2_trigger_init: Board state insufficient for operation (current "FPGA Loaded", requires "Initialized").
```

FIRMWARE:

`bladeRF-cli -f bladeRF_fw_v2.4.0.img`

Output:
```
Flashing firmware...
[INFO @ host/libraries/libbladeRF/src/backend/usb/usb.c:504] Erasing 3 blocks starting at block 0
[INFO @ host/libraries/libbladeRF/src/backend/usb/usb.c:508] Erasing block 2 (100%)...
[INFO @ host/libraries/libbladeRF/src/backend/usb/usb.c:520] Done erasing 3 blocks
[INFO @ host/libraries/libbladeRF/src/backend/usb/usb.c:731] Writing 483 pages starting at page 0
[INFO @ host/libraries/libbladeRF/src/backend/usb/usb.c:736] Writing page 482 (100%)...
[INFO @ host/libraries/libbladeRF/src/backend/usb/usb.c:747] Done writing 483 pages
[INFO @ host/libraries/libbladeRF/src/driver/spi_flash.c:114] Verifying 483 pages, starting at page 0
[INFO @ host/libraries/libbladeRF/src/backend/usb/usb.c:619] Reading 483 pages starting at page 0
[INFO @ host/libraries/libbladeRF/src/backend/usb/usb.c:623] Reading page 482 (100%)...
[INFO @ host/libraries/libbladeRF/src/backend/usb/usb.c:636] Done reading 483 pages
Successfully wrote firmware to flash!
NOTE: A power cycle is required to load the new firmware.
ad9361_init : Unsupported PRODUCT_ID 0x0
ad9361_init : AD936x initialization error
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/rfic_host.c:99] _rfic_host_initialize: ad9361_init(&phy, (AD9361_InitParam *)board_data->rfic_init_params, dev) failed: An unexpected error occurred
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:187] _bladerf2_initialize: dev->board->set_tuning_mode(dev, default_tuning_mode(dev)) failed: An unexpected error occurred
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:468] bladerf2_open: _bladerf2_initialize(dev) failed: An unexpected error occurred
Failed to open device (first available): An unexpected error occurred
```

CzAwe14, 2024-04-23:
hey there, just wondering have you solved this issue yet?

lirrel, 2024-07-14:
I received a new unit from the vendor, probably the previous was physically broken