Nuand/bladeRF #903: Cannot build with -flto

ZeroChaos-, 2023-02-16:
Link time optimization is increasingly common across linux distros, but this fails to build on gentoo when lto is enabled.

I would expect this is easy to replicate with `CFLAGS=-flto -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing`

Gentoo Bug https://bugs.gentoo.org/861872

ZeroChaos-, 2023-02-16:
Min test case to reproduce `CFLAGS="-flto" cmake ..` will show the warning
Min test case to make it a build failure so it doesn't get lost by accident `CFLAGS="-flto -Werror=lto-type-mismatch" cmake ..`

```
[  4%] Linking C shared library ../../output/libbladeRF.so
/home/zero/development/bladeRF/host/libraries/libbladeRF/src/board/bladerf2/common.h:200:25: warning: type of ‘bladerf2_rfic_init_params_fastagc_burst’ does not match original declaration [-Wlto-type-mismatch]
  200 | extern AD9361_InitParam bladerf2_rfic_init_params_fastagc_burst;
      |                         ^
/home/zero/development/bladeRF/fpga_common/src/ad936x_params.c:308:18: note: ‘bladerf2_rfic_init_params_fastagc_burst’ was previously declared here
  308 | AD9361_InitParam bladerf2_rfic_init_params_fastagc_burst = {
      |                  ^
/home/zero/development/bladeRF/fpga_common/src/ad936x_params.c:308:18: note: code may be misoptimized unless ‘-fno-strict-aliasing’ is used
/home/zero/development/bladeRF/host/libraries/libbladeRF/src/board/bladerf2/common.h:199:25: warning: type of ‘bladerf2_rfic_init_params’ does not match original declaration [-Wlto-type-mismatch]
  199 | extern AD9361_InitParam bladerf2_rfic_init_params;
      |                         ^
/home/zero/development/bladeRF/fpga_common/src/ad936x_params.c:20:18: note: ‘bladerf2_rfic_init_params’ was previously declared here
   20 | AD9361_InitParam bladerf2_rfic_init_params = {
      |                  ^
/home/zero/development/bladeRF/fpga_common/src/ad936x_params.c:20:18: note: code may be misoptimized unless ‘-fno-strict-aliasing’ is used
lto-wrapper: warning: using serial compilation of 6 LTRANS jobs
lto-wrapper: note: see the ‘-flto’ option documentation for more information
```

Jannik2099, 2023-03-02:
The definitions of `AD9361_InitParam` differ.

`thirdparty/analogdevicesinc/no-OS/ad9361/sw/ad9361_api.h` : size 528 bytes.
`fpga_common/include/ad936x.h` : size 536 bytes.

diff between small and big (comments removed):
```
88d87
<     uint32_t fagc_lock_level;
170a170,175
>     uint8_t clk_out_drive;
>     uint8_t dataclk_drive;
>     uint8_t data_port_drive;
>     uint8_t clk_out_slew;
>     uint8_t dataclk_slew;
>     uint8_t data_port_slew;
```

ZeroChaos-, 2023-03-29:
How we doing @rghilduta ?  Any chance a fix is on the horizon?

rghilduta, 2025-05-20:
Fixed and tested! https://github.com/Nuand/bladeRF/commit/cd29f93a587c0d9f32b26d41e09f1193d8ab89b3

closed 2025-05-20 by rghilduta