Nuand/bladeRF #906: 2023.03 Installer not yet visible on downloads page

ratzrattillo, 2023-02-28:
The installer for release 2023.02 (https://github.com/Nuand/bladeRF/releases/tag/2023.02) cannot yet be downloaded at https://www.nuand.com/win_installers/

rghilduta, 2023-02-28:
It's on the todo list! Trying to get a few more patches in before rolling a new Windows installer. Stay tuned!

mpumputis, 2023-07-03:
Any tentative release time for this?

lgoix, 2023-07-05:
I've been able to compile for Windows using this tutorial:
https://github.com/Nuand/bladeRF/wiki/Building-libbladeRF-for-Windows

For the part pthreads-win32, I used this release :
https://github.com/GerHobbelt/pthread-win32/releases/tag/version-3.1.0-release

I was able to open it with my Visual Studio 2022 installation:
https://github.com/GerHobbelt/pthread-win32/tree/master/windows/VS2022

zimmer777, 2023-07-26:
I attempted to build it myself, but ran into too many issues. I would really appreciate a new Windows installer.

lgoix, 2023-08-30:
I've juste update my previous comment to add more infos

zimmer777, 2023-09-03:
Thanks so much for posting the more detailed steps! 

I managed to get it to work with libusb v1.0.26. To get that working, I had to modify the directory structure so the dll and lib files were in [LIBUSB_PATH]/MS64/dll/ - the download package had them in libusb-1.0.26-binaries\VS2015-x64\dll. Also, Windows often blocks downloaded dll and lib files, so those had to be unblocked as well.

Markus43, 2024-06-04:
It's been released 2 weeks ago, right?

TobyXue, 2024-11-21:
> * https://github.com/Nuand/bladeRF/tree/41ef63460956e833c9b321252245257ab3946055

Hello lgoix,

Many thanks for this tutorial! However, I set everything the same as yours, but I am encountering the error message of "CMake Error at C:/Program Files/CMake/share/cmake-3.26/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Curses (missing: CURSES_LIBRARY CURSES_INCLUDE_PATH)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.26/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files/CMake/share/cmake-3.26/Modules/FindCurses.cmake:268 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  utilities/bladeRF-power/CMakeLists.txt:6 (find_package)"
  
  Do you have any ideas why is this happening? I also saw another person who has the same issue. 
  
  Thanks in advance!

feraculix, 2025-01-20:
It is impossible to express my gratitude for your detailed compilation guide in words ... working!
I am still struggling to obtain a desired (... I thought a minimalistic/simple one ...) functionality with the libbladeRF API, but being able to compile the API (and especially the relatively nicely working CLI) could bring me a step forward ... eventually, the bladeRF 2.0 might be useful for people not solely relying on gnuradio and similar tools ..

Thank you very much!
(just registered on github for this reply)