Nuand/bladeRF #907: Compiling Error - Cmake pthread - Ubuntu 20.04

RFNOD, 2023-03-01:
Hi,

When compiling the latest version of libbladerf, i get the following error: (I have already verified that I have libpthread-* installed)
This is how i am using cmake:
`cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr/local -DINSTALL_UDEV_RULES=ON ../`

and i get the following error:

`--
CMake Error at libraries/libbladeRF/CMakeLists.txt:428 (cmake_policy):
  Policy "CMP0115" is not known to this version of CMake.

-- libbladeRF version: 2.5.0-git-7e26f5a5
-- nuand bladeRF udev rules will be installed to '/etc/udev/rules.d' upon running 'make install'
--
-- Found libedit: /usr/include, /usr/lib/x86_64-linux-gnu/libedit.so
-- Will regenerate bladeRF-cli interactive help
-- libtecla OFF TRUE
-- libedit OFF TRUE
-- libtecla support enabled
-- Not building man page for bladeRF-cli. (BUILD_BLADERF_CLI_DOCUMENTATION is OFF)
-- Configured to build bladeRF-cli version: 1.9.0-git-7e26f5a5
-- Configuring incomplete, errors occurred!
`
I looked at the cmakeerror.log and this is what i get:

`Performing C SOURCE FILE Test CMAKE_HAVE_LIBC_PTHREAD failed with the following output:
Change Dir: /home/dgs/DGS-SDR/SDRs/bladeRF/host/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_2a5d8/fast && /usr/bin/make -f CMakeFiles/cmTC_2a5d8.dir/build.make CMakeFiles/cmTC_2a5d8.dir/build
make[1]: Entering directory '/home/dgs/DGS-SDR/SDRs/bladeRF/host/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_2a5d8.dir/src.c.o
/usr/bin/cc  -I/usr/include/libusb-1.0  -DCMAKE_HAVE_LIBC_PTHREAD   -o CMakeFiles/cmTC_2a5d8.dir/src.c.o   -c /home/dgs/DGS-SDR/SDRs/bladeRF/host/build/CMakeFiles/CMakeTmp/src.c
Linking C executable cmTC_2a5d8
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_2a5d8.dir/link.txt --verbose=1
/usr/bin/cc  -DCMAKE_HAVE_LIBC_PTHREAD    -rdynamic CMakeFiles/cmTC_2a5d8.dir/src.c.o  -o cmTC_2a5d8
/usr/bin/ld: CMakeFiles/cmTC_2a5d8.dir/src.c.o: in function `main':
src.c:(.text+0x46): undefined reference to `pthread_create'
/usr/bin/ld: src.c:(.text+0x52): undefined reference to `pthread_detach'
/usr/bin/ld: src.c:(.text+0x63): undefined reference to `pthread_join'
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_2a5d8.dir/build.make:87: cmTC_2a5d8] Error 1
make[1]: Leaving directory '/home/dgs/DGS-SDR/SDRs/bladeRF/host/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_2a5d8/fast] Error 2

Source file was:
#include <pthread.h>

void* test_func(void* data)
{
  return data;
}

int main(void)
{
  pthread_t thread;
  pthread_create(&thread, NULL, test_func, NULL);
  pthread_detach(thread);
  pthread_join(thread, NULL);
  pthread_atfork(NULL, NULL, NULL);
  pthread_exit(NULL);

  return 0;
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/dgs/DGS-SDR/SDRs/bladeRF/host/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_576c9/fast && /usr/bin/make -f CMakeFiles/cmTC_576c9.dir/build.make CMakeFiles/cmTC_576c9.dir/build
make[1]: Entering directory '/home/dgs/DGS-SDR/SDRs/bladeRF/host/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_576c9.dir/CheckFunctionExists.c.o
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_576c9.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.16/Modules/CheckFunctionExists.c
Linking C executable cmTC_576c9
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_576c9.dir/link.txt --verbose=1
/usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create    -rdynamic CMakeFiles/cmTC_576c9.dir/CheckFunctionExists.c.o  -o cmTC_576c9  -lpthreads
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_576c9.dir/build.make:87: cmTC_576c9] Error 1
make[1]: Leaving directory '/home/dgs/DGS-SDR/SDRs/bladeRF/host/build/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_576c9/fast] Error 2
`

To test to see if it was my machine, I removed the clone of the master repo and cloned the 2022.11 repo and I was able to compile the older version from source with no problem.

On Ubuntu 20.04
`Linux dgs-V110G2 5.15.0-60-generic #66~20.04.1-Ubuntu SMP Wed Jan 25 09:41:30 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux`

robertghilduta, 2023-03-01:
What version of CMake (cmake --version) are you using?

RFNOD, 2023-03-02:
Hi.. v3.16.3

rghilduta, 2023-03-04:
Fixed in fb1566431789c7c049ad2088682ec0d11fa491f8 !

closed 2023-03-04 by rghilduta

RFNOD, 2023-03-05:
Thank you @rghilduta