Nuand/bladeRF #910: BladeRF sampling frequency for two ports

maurokenny, 2023-03-16:
Hi,

I've seen that the sampling rate for the BladeRF xA4 is 61.44 MHz.
When I activate two ports using python I've seen that each Rx has 30.72 MHz. Is it normal? Is the sampling frequency split?

rghilduta, 2023-03-17:
The sampling rate is not split. Both ports should continue sampling at the requested sampling rate (61.44MHz) independently. How are you setting and reading the sampling rate?

maurokenny, 2023-03-17:
Hi, thanks for the fast reply.

Here you are part of my code:

```
from bladerf import _bladerf

SAMPLE_RATE = 61440000
BANDWIDTH   = 56000000

devices = _bladerf.get_device_list()
bladerf1 = _bladerf.BladeRF(devinfo=devices[0])

def rx_config(bladerf1, rate, bandwidth, gain,
           channel, channel1):
    # activate channels
    ch = bladerf1.Channel(channel)
    ch1 = bladerf1.Channel(channel1)
    # set sample rate
    ch.sample_rate = rate
    ch1.sample_rate = rate
    # set bandwidth
    ch.bandwidth = bandwidth
    ch1.bandwidth = bandwidth

    ch.gain = gain
    ch1.gain = gain

    ch.enable = True
    ch1.enable = True

    return ch, ch1

def rx(bladerf1, num_samples):
    # Setup synchronous stream MIMO 2x2
    bladerf1.sync_config(layout=_bladerf.ChannelLayout.RX_X2,
        fmt=_bladerf.Format.SC16_Q11,
        num_buffers=32,
        buffer_size=32768,
        num_transfers=16,
        stream_timeout=1000)

    bytes_per_sample = 8
    nb_each_16bits = 2
    num_channel = 2

    buf = bytearray(2*num_samples*num_channel*nb_each_16bits)

    if num_samples > 0:
        num = min(len(buf)//bytes_per_sample,
                    num_samples)
    else:
        num = len(buf)//bytes_per_sample
    # Read into buffer
    bladerf1.sync_rx(buf, num)
    data = np.frombuffer(buf[:len(buf)//2], dtype='<i2')

    signal_i = data[0:-3:4] / 2048
    signal_q = data[1:-2:4] / 2048
    signal_i1 = data[2:-1:4] / 2048
    signal_q1 = data[3::4] / 2048
    signal = signal_i + 1j*signal_q
    signal1 = signal_i1 + 1j*signal_q1

    return signal, signal1

channels = rx_config(bladerf1, rate=SAMPLE_RATE, bandwidth=BANDWIDTH, gain=int(0), channel=_bladerf.CHANNEL_RX(0), channel1=_bladerf.CHANNEL_RX(1))
RX1, RX2 = rx(bladerf1, num_samples=int(2**21))
```

I am receiving a WARNING as follow:
`[WARNING @ host/libraries/libbladeRF/src/board/bladerf2/common.c:362] The total sample throughput for the 2 active channels, 122.88 Msps, is greater than the recommended maximum sample throughput, 80 Msps. You may experience dropped samples unless the sample rate is reduced, or some channels are deactivated.`

I would like just to know if I am reading correctly the data and if it is normal to receive this warning.