Nuand/bladeRF #912: issue with setting the samplerate of both RX channels to 61.44M

timelord109, 2023-03-21:
hello

I'm using bladeRF-cli and i'm trying to set the bladeRF sample rate to 61.44M in both RX channels. 
I'm executing the following script
 
`set frequency rx1 2.45G
set frequency rx2 2.45G
set samplerate rx1 61.44M
set samplerate rx2 61.44M
set bandwidth 56M
set bitmode 16
rx config file=fifo.bin format=bin channel=1,2 n=0 timeout=10000
rx start 
rx wait`

I get the following warning:

`[WARNING @ host/libraries/libbladeRF/src/board/bladerf2/common.c:356] The total sample throughput for the 2 active channels, 122 Msps, is greater than the recommended maximum sample throughput, 80 Msps. You may experience dropped samples unless the sample rate is reduced, or some channels are deactivated.
`
i'm using a fifo to plot the samples in the Gnuradio GUI and i get the following results that shows that the signal is corrupted  
[image]

does it mean that we cannot use both rx channel at the same time with a sample rate that is 61Msps each ? 
if it is not the case. is-it possible to do it without changing the bitmode to 8 bits?
N.B: i'm trying not to use new feature where you set the bitmode to 8 bits because it can create issue in term of accuracy and precision

timelord109, 2023-03-21:
i forgot to mention that i'm using a synthétiser to générate a singe tone in both channels input. and that i get the correct result when i set bitmode to 8

rghilduta, 2023-03-31:
2x receive should work without an issue. Can you please retry the bladeRF-cli commands without running "set bitmode". Also, please post the output of "info" in bladeRF-cli, is it reporting "Speed" as "Superspeed" ?

maurokenny, 2023-03-31:
Hi @rghilduta, we haven't use set bitmode and we are in SuperSpeed

```bash
bladeRF> info

  Board:                    Nuand bladeRF 2.0 (bladerf2)
  Serial #:                 9f064e6dea2xxxxxxxxxxxxxxxxxxxxxxxxxxxxx
  VCTCXO DAC calibration:   0x205b
  FPGA size:                49 KLE
  FPGA loaded:              yes
  Flash size:               32 Mbit
  USB bus:                  2
  USB address:              33
  USB speed:                SuperSpeed
  Backend:                  libusb
  Instance:                 0
```

timelord109, 2023-03-31:
* here is the config file we are using 
[script_16bits_channelx1.txt](https://github.com/Nuand/bladeRF/files/11121702/script_16bits_channelx1.txt)
* we use it with bladerf-cli to configure the bladerf . the only thing we change is the  RX channel:
*for 1 RX channel we get the correct complexe single tone response :
[image]
*for 2 RX channels we get this : 
[image]

rthomp10, 2023-04-06:
I'm having trouble reproducing the issue on my end, although there is a chance that the issue is sourced from dropped samples. 

Below is a +15.36MHz CW into RX1 and RX2 with channel 1 and 2 enabled through the cli.

[image]

```
set frequency rx1 2.45G
set frequency rx2 2.45G
set samplerate rx1 61.44M
set samplerate rx2 61.44M
set bandwidth 56M

rx config file=result2.bin format=bin channel=1,2 n=10000
rx start
rx wait
```

In a circumstance where I'm looping back with 2xTX channels to 2xRX channels at 61.44MHz, we'll see the result of USB packet congestion. Dropped samples in this circumstance is expected but serves as a potential root cause indicator.

[image]

```
set loopback rfic_bist
set frequency tx1 2.45G
set frequency tx2 2.45G
set frequency rx1 2.45G
set frequency rx2 2.45G
set samplerate tx1 61.44M
set samplerate tx2 61.44M
set samplerate rx1 61.44M
set samplerate rx2 61.44M
set bandwidth 56M

rx config file=result2.bin format=bin channel=1,2 n=10000 timeout=10000
tx start
rx start
rx wait
tx stop

```

rthomp10, 2026-08-27:
Since my script above only gathered `n=10k` samples, not a continuous stream `n=0`, it dodged RX FIFOs filling in time to witness RX overruns back when we were troubleshooting this. Continuos RX streams will likely start seeing overruns when USB throughput exceeds 80 Msps (320 MB/s).

Consider using  `BLADERF_FORMAT_SC16_Q11_PACKED`. It packs samples as SC12_Q11 in HDL and libbladeRF unpacks to SC16_Q11. If the compute for unpacking is available, it will bump the max throughput by 33%.

In the CLI, `set bitmode packed`.

closed 2026-08-27 by rthomp10