Nuand/bladeRF #913: BLADERF_FORMAT_SC8_Q7_META not working?

manoskav, 2023-03-24:
Hello,
does BLADERF_FORMAT_SC8_Q7_META really work? I am trying to use it, but I get overruns in all bladerf_sync_rx calls. I am running the example [https://github.com/Nuand/bladeRF/blob/master/host/libraries/libbladeRF/doc/examples/sync_rx_meta.c](https://github.com/Nuand/bladeRF/blob/master/host/libraries/libbladeRF/doc/examples/sync_rx_meta.c) where I only change the format in bladerf_sync_config call. Also, the scheduled sync_rx part of the example seems to get stuck.

I am using bladeRF 2.0 micro xA4 on a raspberry pi 4.
Everything is on latest version as seen below from bladeRF-cli:
```
bladeRF> version

  bladeRF-cli version:        1.9.0-git-12699a5c
  libbladeRF version:         2.5.0-git-12699a5c

  Firmware version:           2.4.0-git-a3d5c55f
  FPGA version:               0.15.0 (configured from SPI flash)
```

I get the following output when executing the test:

```
Opening and initializing device...

RX frequency: 1575420000 Hz
RX samplerate: 1000000 sps
RX bandwidth: 1000000 Hz
RX gain: 60
TX frequency: 1575420000 Hz
TX samplerate: 1000000 sps
TX bandwidth: 1000000
TX gain: 60

Running RX meta "now" example...
Overrun detected. 1016 valid samples were read.
Overrun detected. 1016 valid samples were read.
Overrun detected. 1016 valid samples were read.
Overrun detected. 1016 valid samples were read.
Overrun detected. 1016 valid samples were read.
Overrun detected. 1016 valid samples were read.
Overrun detected. 1016 valid samples were read.
Overrun detected. 1016 valid samples were read.
Overrun detected. 1016 valid samples were read.
Overrun detected. 1016 valid samples were read.
Overrun detected. 1016 valid samples were read.
Overrun detected. 1016 valid samples were read.
Overrun detected. 1016 valid samples were read.
Overrun detected. 1016 valid samples were read.
Overrun detected. 1016 valid samples were read.

Running RX meta "scheduled" example...
Current RX timestamp: 0x000000000000448e
[ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x55a77db230
Scheduled RX failed: Operation timed out
```

BLADERF_FORMAT_SC16_Q11_META works as expected.

_Originally posted by @manoskav in https://github.com/Nuand/bladeRF/issues/884#issuecomment-1480396968_

rghilduta, 2023-03-31:
We're getting closer to a solution, please hold tight!

manoskav, 2023-03-31:
> We're getting closer to a solution, please hold tight!

Holding tight! Thanks for the support!

manoskav, 2023-03-31:
Holding tight! Thanks for the support!
(Accidentally closed issue, re-opened now)

manoskav, 2023-03-31:
Holding tight! Thanks!
(accidentally closed issue)

closed 2023-03-31 by rthomp10

rthomp10, 2023-03-31:
The FPGA image will have to be recompiled, but tests are passing! Now you can switch between formats with the `-b <8|16>` within the example.

manoskav, 2023-03-31:
> The FPGA image will have to be recompiled, but tests are passing! Now you can switch between formats with `-b <8|16>` within the example.

Thank you @rthomp10 !
Will the new FPGA image be uploaded at https://www.nuand.com/fpga_images/ ?

rthomp10, 2023-04-03:
Sure thing @manoskav. The updated FPGA image will most likely be up on the site this week.

manoskav, 2023-04-19:
Hello!

Any news on the 8-bit format fix?

rghilduta, 2023-04-19:
The FPGA image for v0.15.0 have been updated on the website. Please give them a try and let us know!

manoskav, 2023-04-19:
Ohh I thought it was not updated due to the older date and version.
I will test and let you know!
Thanks!

manoskav, 2023-05-12:
Everything works as expected now with the SC8Q7_META format! Thanks for the fast support and fixes!

commit aab9d263 (2023-03-31): hdl: timestamp delay added for 8bit format addressing 

Fixes #913