Nuand/bladeRF #915: Problem setting Tuning mode

saxywarrior, 2023-04-06:
I'm trying to enter the command: 

  cal dc rxtx

I get the output: 

  RX DC I: Value = -1792, Error = 0.249
  RX DC Q: Value = -1792, Error = 16.514
  If tuning mode is FPGA, consider setting tuning mode to host by running 'set tuning_mode host'.
  Error: Operation not supported

When I run the set tuning_mode host command it gives no output, and when I run 'cal dc rxtx' again I get teh same output as before

rthomp10, 2023-04-07:
bladeRF 2.0 TX calibration is currently unsupported within the CLI. The current implementation uses the deprecated gain functions from libbladeRF.

closed 2023-04-18 by rthomp10

warnes, 2023-04-18:
Is there any reason not to update the CLI to enable tuning on the bladeRF 2.0?

rghilduta, 2023-04-18:
The bladeRF 2.0 micro’s AD9361 performs all calibrations automatically so the manual method of tuning/calibrating the transceiver is not necessary as it is on the bladeRF 1.0.