Nuand/bladeRF #917: X40   Error (bladerf.script:0): An unexpected error occurred

cn0xroot, 2023-04-14:
`gps-sdr-sim# bladeRF-cli -s bladerf.script 

  Error (bladerf.script:0): An unexpected error occurred

gps-sdr-sim# bladeRF-cli -i
bladeRF> version

  bladeRF-cli version:        1.9.0-git-2b90ed39
  libbladeRF version:         2.5.0-git-2b90ed39

  Firmware version:           2.4.0-git-a3d5c55f
  FPGA version:               0.14.0 (configured from SPI flash)

`

rghilduta, 2023-04-18:
It seems like the PC is aggressively trying to power down XHCI devices. Can you put the governmor into performance mode?
Is this a VM or a physical machine?

Also, what's in bladerf.script?

cn0xroot, 2023-04-18:
1. physical machine

2.
```
cat bladerf.script 
set frequency tx 1575.42M
set samplerate tx 2.6M
set bandwidth tx 2.5M
set gain tx -25
cal lms
cal dc tx
tx config file=gpssim.bin format=bin
tx start
tx wait 
```
and also try default config file: https://github.com/osqzss/gps-sdr-sim/blob/master/bladerf.script

```
For best results, it is not recommended to set both RX and TX to the
same frequency. Instead, consider offsetting them by at least 1 MHz
and mixing digitally.

For the above reason, 'set frequency <value>` is deprecated and
scheduled for removal in future bladeRF-cli versions.

Please use 'set frequency rx' and 'set frequency tx' to configure
channels individually.

  Error (s.script:0): An unexpected error occurred
```

rghilduta, 2023-04-18:
I would suggest removing:
cal lms
cal dc tx
from the script if you're using a bladeRF 2.0 micro

cn0xroot, 2023-04-18:
> I would suggest removing: cal lms cal dc tx from the script if you're using a bladeRF 2.0 micro
thanks for comment

using bladerf X40

cal lms cal dc tx remove,and get the same output:
`  Error (bladerf.script:0): An unexpected error occurred`