Nuand/bladeRF #919: Firmware and FPGA version mismatch on latest release causing warning

jcmartin, 2023-05-06:
Unless I am mistaken, the latest version of the firmware is v2.4.0 (https://www.nuand.com/fx3_images/), but when you load the latest FPGA version (v.0.15.0), you receive a warning message when you using the latest released version of libbladeRF.

[WARNING @ /privatehost/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:165] FPGA v0.15.0 was detected, which requires firmware v2.5.0 or later. The device firmware is currently v2.4.0. Please upgrade the device firmware before continuing.

This has occurred for me under macOS 12.6 and bladeRF 2 A4. However, I imagine that this is a general problem.

jcmartin, 2023-05-06:
I noticed that this issue was resolved in this commit:
0c9b7dfd83bb92b7c11c3b14f3cb70f72a587093

Feel free to close this issue.

closed 2023-05-11 by rthomp10

justacec, 2023-05-14:
Not sure how this resolves the issue.  Looking at the commit, it is setting the minimum FX3 FW to 2.5.0 vice 2.4.0 which is exactly the error that I am getting.  (The same one in the original post)