Nuand/bladeRF #931: (...)bladerf2_open: _bladerf2_initialize(dev) failed: Operation timed out

skyroz, 2023-06-24:
Hello all,

I just want to cut straight to the chase. I am helping a customer of mine to get a variety of spectrum analysis software and hardware running on his computer. We've not had any issues until we started trying to mess with the bladeRF 2.0 Micro A9. This thing--and I'm putting this lightly--is extremely frustrating to get up and running. I am a newbie to spectrum analysis (as I say this is mainly for a customer) but we have not had any other spectrum analyser fail this hard on so many levels. I've tried putting the device into DFU mode and reflashing it, with 4, yes, FOUR different firmwares starting with the most current one v2.4.0 and ending with the first one that was released with this particular model of device, v0.8.0. I've tried 3-4 different FPGAs as well. Nothing--and I mean nothing, can get it to work. My customer is well versed in spectrum analysers as well and even he could not figure it out. Can someone riddle me this? terminal readout for bladeRF-cli coming right at you (and this is all the same regardless of what firmware or fpga combination I've used):

Code: [Select all](https://nuand.com/forums/viewtopic.php?t=13087#)

~$: bladeRF-cli -i
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:122] _bladerf2_initialize: dev->backend->get_fpga_version(dev, &board_data->fpga_version) failed: Operation timed out
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:468] bladerf2_open: _bladerf2_initialize(dev) failed: Operation timed out
Failed to open device (first available): Operation timed out
skyroz666saile@skyroz666saile-MS-7693:~$ bladeRF-cli -i -v verbose
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:642] Using libusb version: 1.0.25.11696
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:519] Found a bladeRF (idx=9)
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/libusb.c:189] Bus 009 Device 002: Nuand bladeRF 2.0, serial 669009ffa96f4fa5a37034dc0cecbd32
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:581] USB port reset succeeded for bladeRF 669009ffa96f4fa5a37034dc0cecbd32
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 0
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:293] Read Firmware version: 2.2.0-git-3d38fac2
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:299] Capability mask before FPGA load: 0x0000000f00000000
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:367] FX3 firmware v2.2.0 does not support SPI flash ID. A firmware update is recommended in order to probe the SPI flash ID information.
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:420] Unknown flash manufacturer ID.
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:427] Could not decode flash manufacturer/device ID and have an unknown FPGA size. Assume default flash architecture.
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:447] SPI flash total size = 32 Mbit
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:448] SPI flash page size = 256 bytes
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:449] SPI flash erase block size = 65536 bytes
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:450] SPI flash number of pages = 16384
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:451] SPI flash number of erase blocks = 64 pages
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 2
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 1
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:420] Unknown flash manufacturer ID.
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:438] Could not decode flash manufacturer/device ID, but found a 301 kLE FPGA. Setting the most probable flash architecture.
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:447] SPI flash total size = 128 Mbit
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:448] SPI flash page size = 256 bytes
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:449] SPI flash erase block size = 65536 bytes
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:450] SPI flash number of pages = 65536
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:451] SPI flash number of erase blocks = 256 pages
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:400] Assumptions were made about the SPI flash architecture! Flash commands may not function as expected.
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 1
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/nios_legacy_access.c:125] Failed to receive NIOS II response: Operation timed out
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/nios_legacy_access.c:214] Failed to read FPGA version[0]: Operation timed out
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:122] _bladerf2_initialize: dev->backend->get_fpga_version(dev, &board_data->fpga_version) failed: Operation timed out
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:468] bladerf2_open: _bladerf2_initialize(dev) failed: Operation timed out
Failed to open device (first available): Operation timed out

if it helps any my client's computer is running the newest version of DragonOS GNU/Linux, I am running Ubuntu 22.04 GNU/Linux, and my client told me when I spoke to him about this device that he could not get it running in windows 10 either. So that's three devices that this has been tested and issue replicated on. I am going to post my issue to as many forums as possible because I've already seen threads about this issue but not seen any satisfactory answers. It's all way too obtuse for me and my client--he's furious with nuand over these issues he's been having with this device, and like I say he is not new to this stuff at all.

rghilduta, 2023-06-24:
FX3 Version 2.2.0 is a very dated release. Please try [v2.4.0](https://www.nuand.com/fx3/bladeRF_fw_v2.4.0.img). Are you running in a virtual machine? If so is XHCI support enabled? It would be interesting to see what is in the dmesg logs when the bladeRF 2.0 is plugged in (please power cycle after flashing FX3 firmware). It also seems like the OTP may have been wiped, if you email your serial number to bladeRF@nuand.com we can look up the board's calibration data.

skyroz, 2023-06-24:
@rghilduta 

Hello and thank you for a swift reply. I will get back to you on the serial number tonight... how would I find the serial number? as i said I couldnt get any firmwares working on here so i will send the serial number as soon as i get back home from one of my jobs this evening.

skyroz, 2023-06-25:
also p.s... no virtual machines here. Just regular computers.

skyroz, 2023-06-27:
> FX3 Version 2.2.0 is a very dated release. Please try [v2.4.0](https://www.nuand.com/fx3/bladeRF_fw_v2.4.0.img). Are you running in a virtual machine? If so is XHCI support enabled? It would be interesting to see what is in the dmesg logs when the bladeRF 2.0 is plugged in (please power cycle after flashing FX3 firmware). It also seems like the OTP may have been wiped, if you email your serial number to [bladeRF@nuand.com](mailto:bladeRF@nuand.com) we can look up the board's calibration data.

hi again. I did send an email to the aformentioned email address and have not heard any reply back so far. i did install the 2.4.0 firmware which got rid of a lot of errors from the terminal readout but still the device will not initialize or open. just thought i would give an update on that.

thanks and have a good day

anton7811, 2023-10-18:
Hi @rghilduta I ran into the same issue. But in my case I have firmware 2.4.0-git-a3d5c55f
I was able to use this device once after unpacked, but the next day I ran in to this problem.
Could you please suggest the next steps to solve/debug the issue?

Here is the full log:
```
$ bladeRF-cli -i -v verbose 
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:642] Using libusb version: 1.0.25.11696
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:519] Found a bladeRF (idx=0)
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/libusb.c:189] Bus 004 Device 010: Nuand bladeRF 2.0, serial c3935ec248074f9c8443f923ee2338b2
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:581] USB port reset succeeded for bladeRF c3935ec248074f9c8443f923ee2338b2
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 0
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:293] Read Firmware version: 2.4.0-git-a3d5c55f
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:299] Capability mask before FPGA load: 0x0000007f00000000
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:401] Found SPI flash manufacturer: WINBOND.
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:409] Found SPI flash device: W25Q128JV (128 Mbit).
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:447] SPI flash total size = 128 Mbit
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:448] SPI flash page size = 256 bytes
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:449] SPI flash erase block size = 65536 bytes
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:450] SPI flash number of pages = 65536
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:451] SPI flash number of erase blocks = 256 pages
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 2
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 3
[DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:447] Loading FPGA from: /usr/share/Nuand/bladeRF/hostedxA4.rbf
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 3
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 1
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/nios_legacy_access.c:125] Failed to receive NIOS II response: Operation timed out
[DEBUG @ host/libraries/libbladeRF/src/backend/usb/nios_legacy_access.c:214] Failed to read FPGA version[0]: Operation timed out
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:122] _bladerf2_initialize: dev->backend->get_fpga_version(dev, &board_data->fpga_version) failed: Operation timed out
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:468] bladerf2_open: _bladerf2_initialize(dev) failed: Operation timed out
Failed to open device (first available): Operation timed out
```

anton7811, 2024-03-31:
Just in case someone has this problem here is a solution: https://github.com/Nuand/bladeRF/commit/a16a6703391ef5c3f19618f0db059e94a31a3c7f

Unfortunately I haven't found this in the documentation.