Nuand/bladeRF #938: Problem occurred while building package uhd: Build failed.

Altinsoy, 2023-09-08:
(.venv) ahmet@ahmet-VirtualBox:~/pybombs$ pybombs prefix init ~/pybombs/bladeRF -a bladeRF -R gnuradio-default
[INFO] Prefix Python version is: 3.11.4
[INFO] PyBOMBS Version 2.3.5
[WARNING] There already is a prefix in `/home/ahmet/pybombs/bladeRF'.
Continue using this path Y/[N]? y
[INFO] Prefix Python version is: 3.11.4
Alias `bladeRF' already exists, overwrite Y/[N]? y
[INFO] Prefix Python version is: 3.11.4
[INFO] Installing default packages for prefix...
[INFO] 
  - gnuradio
[INFO] Install python-apt to speed up apt processing.
[INFO] Phase 1: Creating install tree and installing binary packages:
Install tree:
|
\- gnuradio
   |
   \- uhd
[INFO] Phase 1 complete: All binary dependencies installed.
[INFO] Phase 2: Recursively installing source packages to prefix:
[INFO] Installing package: uhd
[WARNING] A source build for package uhd was requested, but binary install was found!
Install uhd from source despite binary install available Y/[N]? y
[INFO] Install python-apt to speed up apt processing.
[WARNING] Build dir already exists: /home/ahmet/pybombs/bladeRF/src/uhd/host/build
Building:    (100%) [=========================================================]
[  2%] Built target uhd_rpclib
[  5%] Built target uhd-resources
[  6%] Building CXX object lib/CMakeFiles/uhd.dir/usrp/mpmd/sim_find.cpp.o
In file included from /home/ahmet/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/attr.h:11,
                 from /home/ahmet/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/pybind11.h:43,
                 from /home/ahmet/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/embed.h:11,
                 from /home/ahmet/pybombs/bladeRF/src/uhd/host/lib/usrp/mpmd/sim_find.cpp:14:
/home/ahmet/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/cast.h: In function ‘std::string pybind11::detail::error_string()’:
/home/ahmet/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/cast.h:422:36: error: invalid use of incomplete type ‘PyFrameObject’ {aka ‘struct _frame’}
  422 |                 "  " + handle(frame->f_code->co_filename).cast<std::string>() +
      |                                    ^~
In file included from /usr/include/python3.11/Python.h:42,
                 from /home/ahmet/pybombs/bladeRF/src/uhd/host/lib/usrp/mpmd/sim_find.cpp:13:
/usr/include/python3.11/pytypedefs.h:22:16: note: forward declaration of ‘PyFrameObject’ {aka ‘struct _frame’}
   22 | typedef struct _frame PyFrameObject;
      |                ^~~~~~
/home/ahmet/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/cast.h:422:75: error: expected primary-expression before ‘>’ token
  422 |         "  " + handle(frame->f_code->co_filename).cast<std::string>() +
      |                                                                   ^

 /home/ahmet/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/cast.h:422:77: error: expected primary-expression before ‘)’ token
  422 |         "  " + handle(frame->f_code->co_filename).cast<std::string>() +
      |                                                                     ^

/home/ahmet/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/cast.h:424:29: error: invalid use of incomplete type ‘PyFrameObject’ {aka ‘struct _frame’}
  424 |                 handle(frame->f_code->co_name).cast<std::string>() + "\n";
      |                             ^~
/usr/include/python3.11/pytypedefs.h:22:16: note: forward declaration of ‘PyFrameObject’ {aka ‘struct _frame’}
   22 | typedef struct _frame PyFrameObject;
      |                ^~~~~~
/home/ahmet/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/cast.h:424:64: error: expected primary-expression before ‘>’ token
  424 |              handle(frame->f_code->co_name).cast<std::string>() + "\n";
      |                                                             ^

/home/ahmet/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/cast.h:424:66: error: expected primary-expression before ‘)’ token
  424 |              handle(frame->f_code->co_name).cast<std::string>() + "\n";
      |                                                               ^

/home/ahmet/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/cast.h:425:26: error: invalid use of incomplete type ‘PyFrameObject’ {aka ‘struct _frame’}
  425 |             frame = frame->f_back;
      |                          ^~
/usr/include/python3.11/pytypedefs.h:22:16: note: forward declaration of ‘PyFrameObject’ {aka ‘struct _frame’}
   22 | typedef struct _frame PyFrameObject;
      |                ^~~~~~
/home/ahmet/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/pybind11.h: In function ‘pybind11::function pybind11::detail::get_type_override(const void*, const type_info*, const char*)’:
/home/ahmet/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/pybind11.h:2119:49: error: ‘PyThreadState’ {aka ‘struct _ts’} has no member named ‘frame’; did you mean ‘cframe’?
 2119 |     PyFrameObject *frame = PyThreadState_Get()->frame;
      |                                                 ^~~~~
      |                                                 cframe
/home/ahmet/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/pybind11.h:2120:41: error: invalid use of incomplete type ‘PyFrameObject’ {aka ‘struct _frame’}
 2120 |     if (frame && (std::string) str(frame->f_code->co_name) == name &&
      |                                         ^~
/usr/include/python3.11/pytypedefs.h:22:16: note: forward declaration of ‘PyFrameObject’ {aka ‘struct _frame’}
   22 | typedef struct _frame PyFrameObject;
      |                ^~~~~~
/home/ahmet/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/pybind11.h:2121:14: error: invalid use of incomplete type ‘PyFrameObject’ {aka ‘struct _frame’}
 2121 |         frame->f_code->co_argcount > 0) {
      |              ^~
/usr/include/python3.11/pytypedefs.h:22:16: note: forward declaration of ‘PyFrameObject’ {aka ‘struct _frame’}
   22 | typedef struct _frame PyFrameObject;
      |                ^~~~~~
/home/ahmet/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/pybind11.h:2124:18: error: invalid use of incomplete type ‘PyFrameObject’ {aka ‘struct _frame’}
 2124 |             frame->f_locals, PyTuple_GET_ITEM(frame->f_code->co_varnames, 0));
      |                  ^~
/usr/include/python3.11/pytypedefs.h:22:16: note: forward declaration of ‘PyFrameObject’ {aka ‘struct _frame’}
   22 | typedef struct _frame PyFrameObject;
      |                ^~~~~~
In file included from /usr/include/python3.11/Python.h:38:
/home/ahmet/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/pybind11.h:2124:52: error: invalid use of incomplete type ‘PyFrameObject’ {aka ‘struct _frame’}
 2124 |             frame->f_locals, PyTuple_GET_ITEM(frame->f_code->co_varnames, 0));
      |                                                    ^~
/usr/include/python3.11/pyport.h:24:38: note: in definition of macro ‘_Py_CAST’
   24 | #define _Py_CAST(type, expr) ((type)(expr))
      |                                      ^~~~
/usr/include/python3.11/cpython/tupleobject.h:30:38: note: in expansion of macro ‘_PyTuple_CAST’
   30 | #define PyTuple_GET_ITEM(op, index) (_PyTuple_CAST(op)->ob_item[index])
      |                                      ^~~~~~~~~~~~~
/home/ahmet/pybombs/bladeRF/src/uhd/host/lib/deps/pybind11/include/pybind11/pybind11.h:2124:30: note: in expansion of macro ‘PyTuple_GET_ITEM’
 2124 |             frame->f_locals, PyTuple_GET_ITEM(frame->f_code->co_varnames, 0));
      |                              ^~~~~~~~~~~~~~~~
/usr/include/python3.11/pytypedefs.h:22:16: note: forward declaration of ‘PyFrameObject’ {aka ‘struct _frame’}
   22 | typedef struct _frame PyFrameObject;
      |                ^~~~~~
make[2]: *** [lib/CMakeFiles/uhd.dir/build.make:2973: lib/CMakeFiles/uhd.dir/usrp/mpmd/sim_find.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1080: lib/CMakeFiles/uhd.dir/all] Error 2
make: *** [Makefile:166: all] Error 2
[ERROR] Build failed. See output above for error messages.
[ERROR] Problem occurred while building package uhd:
Build failed.
[ERROR] Error installing package uhd. Aborting.

> I tried to install uhd by myself and that time got error about pyzmq. Installed pyzmq but nothing changed. Could you help me to solve this error?

rthomp10, 2023-09-09:
Mind trying [this gr-recipe](https://github.com/rthomp10/gr-recipes)?

Altinsoy, 2023-09-10:
> Mind trying [this gr-recipe](https://github.com/rthomp10/gr-recipes)?

I run the same prefix command after adding the recipes as in the link but I got the same error about uhd.

rthomp10, 2023-09-11:
This is an issue to bring over to the [UHD](https://github.com/EttusResearch/uhd/issues) or [gr-recipes](https://github.com/gnuradio/gr-recipes/issues) repository. Determining which version of UHD that can successfully compile from source will likely determine the [recipe branch target](https://github.com/gnuradio/gr-recipes/blob/master/uhd.lwr#L36).

closed 2023-09-11 by rthomp10