Nuand/bladeRF #939: Documentation for BLADERF_FORMAT_SC8_Q7 is Incorrect

cwozny, 2023-09-10:
The documentation in libbladeRF.h for the new 8-bit format appears to have a copy-paste error:

```
Values in the range [-256, 256) are used to represent [-1.0, 1.0). Note that the lower bound here is inclusive, and the upper bound is exclusive. Ensure that provided samples stay within [-256, 255].

Samples consist of interleaved IQ value pairs, with I being the first value in the pair. Each value in the pair is a right-aligned, little-endian int16_t. The FPGA ensures that these values are sign-extended.
```

It looks like the -256 to 256 is fixed in a future commit after v2.5.0, but the second paragraph talks about each value in the pair being a little-endian 16-bit integer, but I think it should just say: "Each value in the pair is a right-aligned int8_t" since endianness doesn't apply to a byte. I'm also assuming that right-aligned speaks to the ordering of the in-phase value and quadrature value, but maybe that phrase is also unnecessary?

closed 2023-09-11 by rthomp10

rthomp10, 2023-09-11:
Thanks for bringing this up! Change applied. Left in "right-aligned" to clarify the IQ access order within the message.

metasim, 2024-04-26:
@rthomp10 I'm confused how a signed 8 bit value can have the range [-256, 256). Maybe I don't understand what "sign-extended" values are, but I'd like to convert the samples to [-1.0, 1.0), and am flummoxed as to how this should be done.

Also, the name of the encoding is `BLADERF_FORMAT_SC8_Q7` but the first line of the documentation says it's "Signed, Complex 8-bit Q8". Should that be "Signed, Complex 8-bit Q7" instead?

cwozny, 2024-04-26:
Please take a look at the comment in the file on line 2209 and see if your comment is warranted. It was corrected to say -127 to 128.

metasim, 2024-04-26:
My apologies, I was looking at the published documentation. 

However, the source documentation still says `BLADERF_FORMAT_SC8_Q7` is "Signed, Complex 8-bit Q8"

https://github.com/Nuand/bladeRF/blob/2fbae2c38b377cfbee98c281789cd43d1f1b55e4/host/libraries/libbladeRF/include/libbladeRF.h#L2207

commit 07695219 (2023-09-11): doc: corrected value pair type to int8_t

fix #939