Nuand/bladeRF #943: Issues about generating BSP

TobyXue, 2023-09-19:
Hello,

I was trying to measure the code execution time inside of the Nios II, thus an "interval timer" peripheral could be necessary to do this.
Therefore, I recently added an "interval timer" inside of the Nios II system, and this hence lead to some changes upon the corresponding .sopcinfo file. I tried to generate a new BSP regarding the changes in .sopcinfo file via Nios II Eclipse SBT, but Eclipse showed me an error message:
_________
Error executing 'nios2-bsp-generate-files --bsp-dir . --settings settings.bsp', unexpected return code 1.
Error message:
SEVERE: settings.bsp
SEVERE: nios2-bsp-generate-files failed.
_________
[image]

The board I am currently using is: BladeRF2.0 Micro- XA4; OS of the PC: Windows 11. Software versions: Quartus Prime 17.1 and its built-in Eclipse for Nios II SBT. 

Can anyone help? Many thanks in advance.

robertghilduta, 2023-09-19:
What Quartus variant and version number are you using?

TobyXue, 2023-09-20:
Hello Robert, 

Thanks for your reply. I am currently using Quartus Prime 17.1 as suggested.

TobyXue, 2023-09-20:
> What Quartus variant and version number are you using?
> […](#)

Hello Robert,

Thanks for your reply. I am currently using Quartus Prime 17.1 as suggested.

TobyXue, 2023-09-20:
Hello Robert,

Thanks for your reply, I am currently using Qaurtus 17.1 as suggested.

rghilduta, 2023-11-28:
Quartus version is too old, please consider using a more recent one such as >20. Use this script to launch Quartus and compile the BSP: https://github.com/Nuand/bladeRF/blob/master/hdl/quartus/launch_eclipse.sh

closed 2023-11-28 by rghilduta