Nuand/bladeRF #945: Timed out waiting for buf_ready

improperaffo, 2023-09-26:
Hi all,

We are using a project ([srsRAN_Project](https://github.com/srsran/srsRAN_Project)) which only supports UHD, so we are using SoapySDR as a wrapper to control our BladeRF xA9 via UHD. All the commands by UHD (e.g., `uhd_find_devices`, `uhd_usrp_prob`) work, but when we start transmitting via SRS, it seems like the buffers are not ready and we only get these errors:

`[ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:336] wait_for_buffer: Timed out waiting for buf_ready after 1 ms`

We tried to downgrade the BladeRF's FPGA image and FX3 firmware to v0.11.0 (2019-07-31) and v2.3.2 ( 2018-12-16) respectively, but this doesn't solve the problem.

Does anyone have a solution for this problem? Thanks for your help!

Best,
Raphael

leovassalo, 2023-11-18:
Greetings!

Have you got any success on solving this issue?

improperaffo, 2023-11-21:
Yes, we set `static constexpr double RECEIVE_TIMEOUT_S;` and `static constexpr double TRANSMIT_TIMEOUT_S` to `0` (previously they had a value greater than zero.

closed 2023-11-21 by improperaffo