Nuand/bladeRF #949: Pybombs should not be the recommended installing method

SpookyB00k, 2023-10-25:
GNURadio does not recommend using Pybombs anymore for newer versions of GNURadio => 3.9.
https://wiki.gnuradio.org/index.php?title=InstallingGR 

When installing other packages, Pybombs will run in many different problems.
GCC & GG++ should be < 13.
Python should be < 3.11 because UHD uses pybind which will run into a hell lot of trouble when installing.
Building from source or installing via Apt, Dnf or other package managers should be the preferred method.

rghilduta, 2023-11-29:
Good points, this has been a point of discussion! PyBOMBS still offers a fairly reliable way of getting an exact version to run on various platforms. We may end up explicitly version pinning PyBOMBS in the meantime, and in the mid to long term moving it to a backup option.

SpookyB00k, 2023-11-29:
Perhaps add that someone could use update-alternatives for GCC and G++.

However working with pyenv and shims did not work properly. I do not have the solution for that.

I am here not to question the current prefered method of pybombs but could someone monkeypatch a simple warning on the docs regarding the Pybombs installation, that could be of great service. I don't want others to need to troubleshoot this. 

Thank you for your service.

AndreiCherniaev, 2025-09-10:
[Doc](https://github.com/Nuand/bladeRF/wiki/Getting-Started%3A-Linux#easy-installation-pybombs) should be fixed: instead "Easy installation: PyBOMBS" should be something like "Deprecated way of installation: PyBOMBS".