Nuand/bladeRF #950: Error on generating .rbf file

NIDHIPANDA, 2023-11-01:
Hello, I am using Quartus 17.1 version on ubuntu 20.04 LTS system. While regenerating .rbf file I am facing the following issue:

I used command " ./build_bladerf.sh -b bladeRF-micro -r hosted -s A4". It worked up to  

[100%] Built target ad936x  and than gave error 

Info: Linking ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/bladeRF_nios.elf
nios2-elf-g++  -T'../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios_bsp/linker.x' -msys-crt0='../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios_bsp/obj/HAL/src/crt0.o' -msys-lib=hal_bsp -L../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios_bsp -L../../../../../quartus/work/bladerf-micro-A4-hosted/libad936x  -msmallc  -Wl,-Map=../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/bladeRF_nios.map -DNUAND_MODIFICATIONS  -Os -fshort-enums -g -Wall -Wextra -Wno-unused-parameter -Werror  -DBLADERF_NIOS_BUILD -mno-hw-div -mno-hw-mul -mno-hw-mulx -mgpopt=global -DBOARD_BLADERF_MICRO -o ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/bladeRF_nios.elf ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/obj/default/./src/bladeRF_nios.o ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/obj/default/ad936x_helpers.o ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/obj/default/bladerf2_common.o ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/obj/default/range.o ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/obj/default/devices.o ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/obj/default/devices_rfic.o ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/obj/default/devices_rfic_cmds.o ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/obj/default/devices_rfic_queue.o ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/obj/default/devices_sim.o ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/obj/default/pkt_16x64.o ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/obj/default/pkt_32x32.o ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/obj/default/pkt_8x16.o ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/obj/default/pkt_8x32.o ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/obj/default/pkt_8x64.o ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/obj/default/pkt_8x8.o ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/obj/default/pkt_legacy.o ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/obj/default/pkt_retune2.o -lm -lad936x 
nios2-elf-insert ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/bladeRF_nios.elf --thread_model hal --cpu_name nios2 --qsys true --simulation_enabled false --stderr_dev jtag_uart --stdin_dev jtag_uart --stdout_dev jtag_uart --sopc_system_name nios_system --quartus_project_dir "/home/cipl/Documents/Cyronics/codes/bladeRF/hdl/fpga/platforms/bladerf-micro/software/bladeRF_nios" --sopcinfo /home/cipl/Documents/Cyronics/codes/bladeRF/hdl/quartus/work/bladerf-micro-A4-hosted/bladeRF_nios_bsp/../nios_system.sopcinfo
Info: (../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/bladeRF_nios.elf) 121 KBytes program size (code + initialized data).
Info:                                                                                     3572 Bytes free for stack + heap.
Post-processing to create ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/mem_init/nios_system_ram.hex...
elf2hex ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/bladeRF_nios.elf 0x00020000 0x0003ffff --width=32 --record=4 --record=4 --little-endian-mem --create-lanes=0 ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/mem_init/nios_system_ram.hex
Post-processing to create ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/mem_init/nios_system_vctcxo_tamer_0.hex...
elf2hex ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/bladeRF_nios.elf 0x00009300 0x000093ff --width=8 --record=1 --record=1 --little-endian-mem --create-lanes=0 ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/mem_init/nios_system_vctcxo_tamer_0.hex
Post-processing to create ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/mem_init/hdl_sim/nios_system_ram.dat...
elf2dat --infile=../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/bladeRF_nios.elf --outfile=../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/mem_init/hdl_sim/nios_system_ram.dat \
	--base=0x00020000 --end=0x0003ffff --width=32 \
	--little-endian-mem --create-lanes=0 
/opt/quartus_17.1/nios2eds/bin/elf2dat: 2: .: Can't open /bin/sh_pl.sh
make: *** [../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios_bsp/mem_init.mk:292: ../../../../../quartus/work/bladerf-micro-A4-hosted/bladeRF_nios/mem_init/hdl_sim/nios_system_ram.dat] Error 127

Please suggest what went wrong here.  File elf2dat and sh_pl.sh are present in there respective paths.

rghilduta, 2023-11-01:
Quartus 17 may be too out of date, please try at least 18 or more favorably 20+.

NIDHIPANDA, 2023-11-02:
Thank you for your reply. Tried the same with Quartus Version=22.1.2.922 still getting the same error. To set environment variable I am using following command:
export QUARTUS_ROOTDIR=/opt/Quartus/quartus
. $QUARTUS_ROOTDIR/adm/qenv.sh
export PATH=$PATH:/opt/Quartus/quartus/sopc_builder/bin/:/opt/Quartus/nios2eds/sdk2/bin:/opt/Quartus/nios2eds/bin/gnu/H-x86_64-pc-linux-gnu/bin:/opt/Quartus/nios2eds/bin:/opt/Quartus/quartus/bin

Is it that I am missing anything to add. What can be cause of this error.

closed 2023-11-28 by rghilduta