Nuand/bladeRF #956: fx3_gpif entity error

feyzaozdemir, 2023-11-30:
Hi,
I'm trying to build the project by following bladeRF/hdl/readme.md.

rghilduta, 2023-11-30:
Do you see any RBF files in output_files?

feyzaozdemir, 2023-11-30:
No, there are rpt, done, smsg, pin and summary files only

rghilduta, 2023-11-30:
Is there a .sof?

feyzaozdemir, 2023-11-30:
no

feyzaozdemir, 2023-11-30:
I started Analysis&Synthesis and it finished with no errors

feyzaozdemir, 2023-11-30:
I did and didnt generated any of.

feyzaozdemir, 2023-12-01:
What do you mean by the hosted variant?

rghilduta, 2023-12-01:
When you launch ./build_bladerf.sh the -r argument specifies the revision/variant: e.g. adsb or hosted. Just want to make sure hosted can generate an RBF successfully.

feyzaozdemir, 2023-12-01:
No, it generated these files;
hosted.asm.rpt  hosted.fit.smsg     hosted.map.rpt      hosted.pin
hosted.done     hosted.fit.summary  hosted.map.smsg     hosted.sta.rpt
hosted.fit.rpt  hosted.flow.rpt     hosted.map.summary  hosted.sta.summary

feyzaozdemir, 2023-12-01:
Didnt generated again but i saw warning about my quartus version I'll check that

feyzaozdemir, 2023-12-01:
Ok, problem was quartus. Thank you for all your help and time.

closed 2023-12-01 by feyzaozdemir