Nuand/bladeRF #962: Synchronization of TX and RX Signals in BladeRF 2.0 xA9

patilraj1707, 2024-04-18:
I want to implement a FMCW or a pulsed radar using a SDR I tried using Pluto sdr but that does not have any synchronization mechanisms, I do have a BladeRF in my laboratory and I saw that in the API to control the bladeRF there is a bladerf_sync_config(), bladerf_sync_rx(),  bladerf_sync_tx() functions. So, I wanted to know that if I use these functions in my code then will the TX and the RX signals be tine synchronized.

venky0108, 2024-04-18:
on the same board, it can be time synchronized by using trigger functionality of bladerf. Making Tx channel master & Rx channel slave. When trigger is fired, sampling starts.

patilraj1707, 2024-04-18:
is there a sample code available to implement the functionality described by you. Because I am not familiar with the API for bladerf and I am in a bit of a hurry so can't waste much time figuring it out. The connection and the transmission part I can handle but the syncing I need help for that.

Heshyo, 2024-04-19:
I'm not sure what you need, but if you're using only 1 SDR, the timestamp used by the RX and TX is the same.

venky0108, 2024-04-19:
Hey Heshyo, does that mean both TX & RX DAC & DAC start the conversion at same time?

patil bladerf trigger functionality is not available in python binding of libbladerf which is the library we import in python. As per my knowledge trigger functionality is for multi board synchronisation

Heshyo, 2024-04-19:
@venky0108 I'm not sure what you mean by "DAC & ADC start the conversion at the same time", but I expect DAC & ADC to work at the same speed, and for each new sample they update their timestamp by +1. But as the timestamps of RX and TX are expected to be the same, the FPGA code most probably only keeps track of 1 timestamp.

FYI, when you set the sampling rate of RX, that also sets the sampling rate of TX to the same value, and vice versa. So both always end up having the same value.

Also, if you stop streaming on both RX and TX, then you "lose" timestamp continuity, meaning that if you start streaming again exactly 1s later, the new timestamp won't correspond to the previous timestamp +1s. The new timestamp may be reset to 0, or continue from the previous one, or something else. This is because the timestamp is incremented with each new sample.

Heshyo, 2024-04-19:
@venky0108 Yes in dual channel, whether RX or TX, both channel use the same timestamp.