Nuand/bladeRF #971: Problem instaling  Mac OSX

Zadvornyi, 2024-06-04:
I have tried to install compiling from scratch, and everything is installed correctly without error, however stranger and gqrx. I have got errors

[image]

My machine:
[image]

rthomp10, 2024-06-04:
After plugging in the bladeRF, can the bladeRF-cli find it?

```bash
$ bladeRF-cli --probe
```

If not, can it be found using `lsusb`?

```bash
$ lsusb | grep Nuand
Bus 003 Device 065: ID 2jf0:5240 Nuand LLC bladeRF 2.0 micro
```

Zadvornyi, 2024-06-04:
Hello,
[image]

rthomp10, 2024-06-04:
Try setting the device string to `bladerf=0`. This should open the first device libbladeRF can find. More details can be found in [Gnuradio OsmoSDR device string options](https://github.com/Nuand/bladeRF/wiki/Gnuradio-OsmoSDR-device-string-options) in the wiki.

[image]

Zadvornyi, 2024-06-04:
Do you use gqrx?

Zadvornyi, 2024-06-04:
Have you ever try to ru. BladeRF on SDRAngel MacOS?

Zadvornyi, 2024-06-06:
I have set:
[image]
I have got
[image]

On Windows machines, I have used sdrangel this work ok

Zadvornyi, 2024-06-06:
I have fixed the problem. The steps that I have done
- I have download image from [bladeRF_fw_latest.img](https://www.nuand.com/fx3_images/)
- run `bladeRF-cli -f ./bladeRF_fw_latest.img`
- I have download FPGA images from [hostedxA4-latest.rbf](https://www.nuand.com/fpga_images/)
- run ` bladeRF-cli -l ./hostedxA4-latest.rbf`
- run ` bladeRF-cli -e info -e version`

[image]

rthomp10, 2024-06-06:
Nice! The fpga image (.rbf) will have to be loaded every time the bladeRF loses power. 

If you would like to store the fpga image on the bladeRF, try `bladeRF-cli -L hostedxA4-latest.rbf` as this will flash the fpga image to its spi flash. It will then automatically load on power up. `bladeRF-cli -L X` will erase the stored fpga image if the functionality is not preferred.

closed 2024-06-07 by rthomp10