Nuand/bladeRF #972: Fix Argument Order in calloc Call in fir_filter.c

goatmanic, 2024-06-17:
The calloc function call in the fir_filter.c file uses the arguments in an incorrect order, causing a compilation error with the message:

**error: ‘calloc’ sizes specified with ‘sizeof’ in the earlier argument and not in the later argument [-Werror=callc- transposed-args]**

This error occurs because calloc expects the number of elements as the first argument and the size of each element as the second argument. However, the existing code has these reversed.
Affected File

    Path: bladeRF/host/utilities/bladeRF-fsk/c/src/fir_filter.c
    Line: 227

Proposed Fix

The arguments to calloc should be reordered to correctly reflect the expected signature of calloc(num_elements, element_size). The corrected line of code is:

**outbuf = calloc(chunk_size, sizeof(struct complex_sample));**

This change ensures that the chunk_size (the number of elements) is passed as the first argument and the sizeof(struct complex_sample) (the size of each element) is passed as the second argument.
Steps to Reproduce

    Compile the affected file with a strict compiler that treats warnings as errors.
    Observe the compilation error due to the transposed arguments in the calloc function.

bezborodow, 2024-09-22:
I have the same error. Applying this patch allows compilation to succeed. I am using Fedora 40.

rthomp10, 2026-08-28:
Fixed since bladeRF-fsk commit [78a0b10](https://github.com/Nuand/bladeRF-fsk/commit/78a0b106404ab4ba5fb7b8b83e859c5f5517f32a#diff-3b37f906f6a6c53f838cf1dd8f1c17cf9f114c1426cec0c9247151972f0a05eaL225)

closed 2026-08-28 by rthomp10