Nuand/bladeRF #977: Unstable bladeRF on lastest versions

ccgauche, 2024-07-22:
I tested on multiple bladeRFs and I always get errors on my raspberry pi 5
```
Jul 22 15:20:16 pi sh[54887]: [ERROR @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:68] Failed to send NIOS II request: File or device I/O failure
Jul 22 15:20:19 pi sh[54887]: [ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x55561e602920
Jul 22 15:20:22 pi sh[54887]: [124B blob data]
Jul 22 15:20:22 pi sh[54887]: [ERROR] BladeRF error: BladeRFError(Timeout) at crates/sensor/src/lib.rs:344
Jul 22 15:20:22 pi sh[54887]: [ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x55561e56cdf0
Jul 22 15:20:22 pi sh[54887]: [111B blob data]
Jul 22 15:20:22 pi sh[54887]: [111B blob data]
Jul 22 15:20:22 pi sh[54887]: [111B blob data]
Jul 22 15:20:22 pi sh[54887]: [111B blob data]
Jul 22 15:20:22 pi sh[54887]: [111B blob data]
Jul 22 15:20:22 pi sh[54887]: [111B blob data]
Jul 22 15:20:22 pi sh[54887]: [111B blob data]
Jul 22 15:20:23 pi sh[54887]: [28B blob data]
Jul 22 15:20:23 pi sh[54887]: [INFO] No update available
Jul 22 15:20:25 pi sh[54887]: [ERROR @ host/libraries/libbladeRF/src/streaming/sync.c:352] wait_for_buffer: Timed out waiting for buf_ready after 3000 ms
Jul 22 15:20:25 pi sh[54887]: thread 'main' panicked at crates/sensor/src/lib.rs:344:21:
Jul 22 15:20:25 pi sh[54887]: BladeRF error: BladeRFError(Timeout) at crates/sensor/src/lib.rs:344
Jul 22 15:20:25 pi sh[54887]: note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Jul 22 15:20:25 pi sh[54887]: [ERROR @ host/libraries/libbladeRF/src/backend/usb/libusb.c:1089] Transfer timed out for buffer 0x55561e56cdf0
Jul 22 15:20:25 pi sh[54887]: [111B blob data]
Jul 22 15:20:25 pi sh[54887]: [111B blob data]
Jul 22 15:20:25 pi sh[54887]: [111B blob data]
Jul 22 15:20:25 pi sh[54887]: [111B blob data]
Jul 22 15:20:25 pi sh[54887]: [111B blob data]
Jul 22 15:20:25 pi sh[54887]: [111B blob data]
Jul 22 15:20:25 pi sh[54887]: [111B blob data]
Jul 22 15:20:25 pi sh[54887]: [184B blob data]
Jul 22 15:20:25 pi sh[54887]: Write Error -5
```
Or this one:
```
Jul 22 13:56:24 pi sh[34793]: [ERROR @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:77] Failed to receive NIOS II response: File or device I/O failure
Jul 22 13:56:24 pi sh[34793]: [ERROR @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1934] bladerf2_set_correction: ad9361_spi_write( phy->spi, ad9361_correction_rx_dcoff_reg_table[ch][low_band][is_q].reg_top, data_top) failed: An unexpected error occurred
```

Did anyone get the same issue ?

ur3vke, 2024-08-24:
I have the same issue

ccgauche, 2024-08-27:
```
[ERROR @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:77] Failed to receive NIOS II response: File or device I/O failure
[ERROR @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1969] bladerf2_set_correction: ad9361_spi_write(phy->spi, reg, data) failed: An unexpected error occurred
[ERROR] [1724744970] [crates/sensor/src/lib.rs:172] BladeRF error: BladeRFError(Unexpected) at crates/sensor/src/lib.rs:172
```
This error also happens

rghilduta, 2024-08-27:
This is a bug with many causes. On the RPi5 I would suggest trying to power the bladeRF separately via the DC barrel connector.

@ccgauche If you are in VM, please ensure xHCI accleration/passthrough is enabled.

@Andrii067 With regards to buffers dropping, it is imperative that the thread calling bladerf_sync_rx() receives samples then saves them and immediately comes back to calling bladerf_sync_rx(). DSP should be left for another thread.

ccgauche, 2024-08-27:
@rghilduta I also tried with the blade powered from the barrel jack and the error still happens... I'm running flat raspbian on my RPi5... 
```
bladeRF> print

  RX1 Bandwidth:  18000000 Hz (Range: [200000, 56000000])
  RX2 Bandwidth:  18000000 Hz (Range: [200000, 56000000])
  TX1 Bandwidth:  18000000 Hz (Range: [200000, 56000000])
  TX2 Bandwidth:  18000000 Hz (Range: [200000, 56000000])

  RX1 Frequency: 2400000000 Hz (Range: [70000000, 6000000000])
  RX2 Frequency: 2400000000 Hz (Range: [70000000, 6000000000])
  TX1 Frequency: 2400000000 Hz (Range: [47000000, 6000000000])
  TX2 Frequency: 2400000000 Hz (Range: [47000000, 6000000000])

  Tuning Mode: Host

  Bit Mode: 16 bit samples

  Feature:  Default enabled

  RX1 AGC: Enabled   
  RX2 AGC: Enabled   

  Clock reference: none
  Clock input:     Onboard VCTCXO
  Clock output:    Disabled

  RX1 RSSI: preamble = 0 dB, symbol = -52 dB
  RX2 RSSI: preamble = 0 dB, symbol = -52 dB

  Loopback mode: none

  RX mux: BASEBAND - Baseband samples

  RX FIR Filter: normal (default)
  TX FIR Filter: bypass (default)

  Gain RX1 overall:   60 dB (Range: [-15, 60])
              full:   71 dB (Range: [-4, 71])
  Gain RX2 overall:   60 dB (Range: [-15, 60])
              full:   71 dB (Range: [-4, 71])
  Gain TX1 overall:   56 dB (Range: [-23.75, 66])
               dsa:  -90 dB (Range: [-89.75, 0])
  Gain TX2 overall:   56 dB (Range: [-23.75, 66])
               dsa:  -90 dB (Range: [-89.75, 0])

  RX1 sample rate: 30720000 0/1 (Range: [520834, 61440000])
  RX2 sample rate: 30720000 0/1 (Range: [520834, 61440000])
  TX1 sample rate: 30720000 0/1 (Range: [520834, 61440000])
  TX2 sample rate: 30720000 0/1 (Range: [520834, 61440000])

  Bias Tee (RX1): off
  Bias Tee (RX2): off
  Bias Tee (TX1): off
  Bias Tee (TX2): off

  Current VCTCXO trim: 0x1d23
  Stored VCTCXO trim:  0x1d23

  Hardware status:
    RFIC status:
      Tuning Mode:  Host
      Temperature:  37.7 degrees C
      CTRL_OUT:     0xf8 (0x035=0x00, 0x036=0xff)
    Power source:   DC Barrel
    Power monitor:  4.844 V, 0.55 A, 2.66 W
    RF routing:
      RX1: RFIC 0x0 (A_BAL  ) <= SW 0x0 (OPEN   )
      RX2: RFIC 0x0 (A_BAL  ) <= SW 0x0 (OPEN   )
      TX1: RFIC 0x0 (TXA    ) => SW 0x0 (OPEN   )
      TX2: RFIC 0x0 (TXA    ) => SW 0x0 (OPEN   )
```
On the jack barrel.

ccgauche, 2024-09-04:
Edit:
The issue doesn't arises on x86 computers nor the khadas VIM3 ARM board running the same exact software stack as the RPi5. Thus either pointing toward an issue with the USB Controller on the RPi or the Power delivery on the RPi USB ports. Through the issue also happens while plugged with the DC Barrel (maybe an issue with my power cord?)

xdevelnet, 2024-12-27:
@ccgauche have you solved this issue?

ccgauche, 2025-01-06:
@xdevelnet No the issue seem to be unavoidable on Rpi5. Seem stable on the Rpi4 and khadas vim3 board.

rghilduta, 2025-05-11:
Lots of stability improvements available in libbladeRF v2.6.0 https://github.com/Nuand/bladeRF/commit/09c82087786702d2812f3f0fc909beccf3277233

closed 2025-05-11 by rghilduta

ArhumJain, 2025-05-17:
Running libbladeRF v2.6.0 on a Raspberry Pi 5 and this issue persists. Perhaps the issue should be reopened?

xdevelnet, 2025-05-19:
@ArhumJain 

I've seen similar issues on different arm platform.

I currently havent rpi 5, but can you try the following?

1. Use bladerf in usb 3.0 mode
2. Control rpi 5 remotely, via ethernet for example
3. Unplug each usb device that is connected to rpi 5, leave only bladerf connected
4. (optional) use rt kernel