Nuand/bladeRF #978: Modify device serial number

liuyueboye, 2024-07-23:
How to modify bladerf 2.0 micro device serial number？ Can it be modified through bladeRF-cli, or through the fx3 tool?

Heshyo, 2024-07-23:
I doubt there's an API to change the serial number, as it's supposed to be unique.

What would be the use case of changing it?

rghilduta, 2024-07-23:
The serial number is not allowed to be tampered with. It is in One Time Programmable memory. There is no API to modify the device serial.

closed 2024-07-23 by rghilduta

liuyueboye, 2024-07-24:
@Heshyo @rghilduta Thank you both for your replies，It is hoped that different hardware devices can be distinguished by serial numbers. Since the chip was replaced and the fx3 firmware was re-burned, the serial numbers are all  "00000000......".Are there any other ways to modify the serial number? For example, replace the chip and recompile the fx3 firmware? or related tools