Nuand/bladeRF #990: GUI CMake bladerf project generation on windows 11 problem

tarasii, 2024-11-06:
Hi

Windows 11
https://github.com/Nuand/bladeRF.git cloned.
libusb and phtreads-w32 downloaded
CMake 3.3.10 GUI - Configure - and getting:
CMake Error at C:/Program Files/CMake/share/cmake-3.31/Modules/FindPackageHandleStandardArgs.cmake:233 (message):
  Could NOT find Curses (missing: CURSES_LIBRARY CURSES_INCLUDE_PATH)
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.31/Modules/FindPackageHandleStandardArgs.cmake:603 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files/CMake/share/cmake-3.31/Modules/FindCurses.cmake:268 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  utilities/bladeRF-power/CMakeLists.txt:6 (find_package)

Can you please point me how can I install CURSES libruary under windows

Thanks

TobyXue, 2024-11-21:
Hello,

I am also encountering the same issue, but not sure if you followed steps recommended by others, you may refer to this post: https://github.com/Nuand/bladeRF/issues/906 However, I am still encountering the same issue though I followed the recommended steps...

TobyXue, 2024-12-02:
Hello,

After multiple failures, I finally made them to build successfully. I shall explain some essentail steps, for CURSES library under windows, you may refer to this link of pdcurses: https://pdcurses.org/ but I don't think it makes any further efforts for building solutions of libbladeRF, as you may encounter more issues while building, such as couldn't find bladeRF.lib file, or lack of unistd.h etc...

I strongly recommend to follow the steps mentioned in: https://github.com/Nuand/bladeRF/issues/906 strictly, and to git clone this version of repo: https://github.com/Nuand/bladeRF/tree/41ef63460956e833c9b321252245257ab3946055 for building the whole solution of libbladeRF. Then everything should work properly.

feraculix, 2025-01-20:
> I strongly recommend to follow the steps mentioned in: [#906](https://github.com/Nuand/bladeRF/issues/906) strictly, and to git clone this version of repo: [...]

Awesome, thank you very very much for sharing this guide/link!
It finally allowed my to get this stuff compiled on Windows 11, bladeRF-win-installer-2024.05, CMake 3.30.3, PortableGit-2.47.1.2-64-bit, libusb-1.0.27, VS 17.10.5 (Community 2022), pthread from the link you provided. 

Finally the next step into getting the bladeRF 2.0 doing something useful ...