Nuand/bladeRF #991: Installation with PyBOMBS fails

CeSiumUA, 2024-11-21:
Hi!
I've tried to install bladeRF host libraries via PyBOMBS as described here: https://github.com/Nuand/bladeRF/wiki/Getting-Started%3A-Linux#user-content-Easy_installation_PyBOMBS
However, pybombs prefix init ~/pybombs/bladeRF -a bladeRF -R gnuradio-default fails with the following error:

```
[  0%] Building CXX object gnuradio-runtime/lib/pmt/CMakeFiles/gnuradio-pmt.dir/pmt_unv.cc.o
In file included from /home/fedir/pybombs/bladeRF/include/volk/volk.h:19,
                 from /home/fedir/pybombs/bladeRF/include/volk/volk_alloc.hh:18,
                 from /home/fedir/pybombs/bladeRF/src/gnuradio/gnuradio-runtime/lib/pmt/pmt_unv_int.h:16,
                 from /home/fedir/pybombs/bladeRF/src/gnuradio/gnuradio-runtime/lib/pmt/pmt_unv.cc:18:
/home/fedir/pybombs/bladeRF/include/volk/volk_common.h:176:23: error: exponent has no digits
  176 |     const float a1 = +0x1.ffffeap-1f;
      |                       ^~~~~~~~~~~
/home/fedir/pybombs/bladeRF/include/volk/volk_common.h:177:23: error: exponent has no digits
  177 |     const float a3 = -0x1.55437p-2f;
      |                       ^~~~~~~~~~
/home/fedir/pybombs/bladeRF/include/volk/volk_common.h:178:23: error: exponent has no digits
  178 |     const float a5 = +0x1.972be6p-3f;
      |                       ^~~~~~~~~~~
/home/fedir/pybombs/bladeRF/include/volk/volk_common.h:179:23: error: exponent has no digits
  179 |     const float a7 = -0x1.1436ap-3f;
      |                       ^~~~~~~~~~
/home/fedir/pybombs/bladeRF/include/volk/volk_common.h:180:23: error: exponent has no digits
  180 |     const float a9 = +0x1.5785aap-4f;
      |                       ^~~~~~~~~~~
/home/fedir/pybombs/bladeRF/include/volk/volk_common.h:181:24: error: exponent has no digits
  181 |     const float a11 = -0x1.2f3004p-5f;
      |                        ^~~~~~~~~~~
/home/fedir/pybombs/bladeRF/include/volk/volk_common.h:182:24: error: exponent has no digits
  182 |     const float a13 = +0x1.01a37cp-7f;
      |                        ^~~~~~~~~~~
/home/fedir/pybombs/bladeRF/include/volk/volk_common.h: In function ‘float volk_arctan_poly(float)’:
/home/fedir/pybombs/bladeRF/include/volk/volk_common.h:176:35: error: unable to find numeric literal operator ‘operator""f’
  176 |     const float a1 = +0x1.ffffeap-1f;
      |                                   ^~
/home/fedir/pybombs/bladeRF/include/volk/volk_common.h:176:35: note: use ‘-fext-numeric-literals’ to enable more built-in suffixes
/home/fedir/pybombs/bladeRF/include/volk/volk_common.h:177:34: error: unable to find numeric literal operator ‘operator""f’
  177 |     const float a3 = -0x1.55437p-2f;
      |                                  ^~
/home/fedir/pybombs/bladeRF/include/volk/volk_common.h:177:34: note: use ‘-fext-numeric-literals’ to enable more built-in suffixes
/home/fedir/pybombs/bladeRF/include/volk/volk_common.h:178:35: error: unable to find numeric literal operator ‘operator""f’
  178 |     const float a5 = +0x1.972be6p-3f;
      |                                   ^~
/home/fedir/pybombs/bladeRF/include/volk/volk_common.h:178:35: note: use ‘-fext-numeric-literals’ to enable more built-in suffixes
/home/fedir/pybombs/bladeRF/include/volk/volk_common.h:179:34: error: unable to find numeric literal operator ‘operator""f’
  179 |     const float a7 = -0x1.1436ap-3f;
      |                                  ^~
/home/fedir/pybombs/bladeRF/include/volk/volk_common.h:179:34: note: use ‘-fext-numeric-literals’ to enable more built-in suffixes
/home/fedir/pybombs/bladeRF/include/volk/volk_common.h:180:35: error: unable to find numeric literal operator ‘operator""f’
  180 |     const float a9 = +0x1.5785aap-4f;
      |                                   ^~
/home/fedir/pybombs/bladeRF/include/volk/volk_common.h:180:35: note: use ‘-fext-numeric-literals’ to enable more built-in suffixes
/home/fedir/pybombs/bladeRF/include/volk/volk_common.h:181:36: error: unable to find numeric literal operator ‘operator""f’
  181 |     const float a11 = -0x1.2f3004p-5f;
      |                                    ^~
/home/fedir/pybombs/bladeRF/include/volk/volk_common.h:181:36: note: use ‘-fext-numeric-literals’ to enable more built-in suffixes
/home/fedir/pybombs/bladeRF/include/volk/volk_common.h:182:36: error: unable to find numeric literal operator ‘operator""f’
  182 |     const float a13 = +0x1.01a37cp-7f;
      |                                    ^~
/home/fedir/pybombs/bladeRF/include/volk/volk_common.h:182:36: note: use ‘-fext-numeric-literals’ to enable more built-in suffixes
make[2]: *** [gnuradio-runtime/lib/pmt/CMakeFiles/gnuradio-pmt.dir/build.make:76: gnuradio-runtime/lib/pmt/CMakeFiles/gnuradio-pmt.dir/pmt_unv.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:3931: gnuradio-runtime/lib/pmt/CMakeFiles/gnuradio-pmt.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
[ERROR] Build failed. See output above for error messages.
[ERROR] Problem occurred while building package gnuradio:
Build failed.
[ERROR] Error installing package gnuradio. Aborting.
```

OS used: Ubuntu 24.04.1 LTS via WSL

marco-bv, 2024-12-10:
Dear @CeSiumUA,

I had the same issue. The way that I figure to solve was to insert the following line
`set(CMAKE_CXX_STANDARD 17)`

In the following CMakeLists.txt files:

pybombs/bladeRF/src/gnuradio/gnuradio-runtime/lib/pmt/CMakeLists.txt
pybombs/bladeRF/src/gnuradio/gnuradio-runtime/lib/CMakeLists.txt
pybombs/bladeRF/src/gnuradio/gr-blocks/lib/CMakeLists.txt
pybombs/bladeRF/src/gnuradio/gr-blocks/python/blocks/CMakeLists.txt
pybombs/bladeRF/src/gnuradio/gr-blocks/python/blocks/bindings/CMakeLists.txt
pybombs/bladeRF/src/gnuradio/gr-fec/lib/CMakeLists.txt
pybombs/bladeRF/src/gnuradio/gr-fec/python/fec/bindings/CMakeLists.txt
pybombs/bladeRF/src/gnuradio/gr-fft/lib/CMakeLists.txt
pybombs/bladeRF/src/gnuradio/gr-filter/lib/CMakeLists.txt
pybombs/bladeRF/src/gnuradio/gr-analog/lib/CMakeLists.txt
pybombs/bladeRF/src/gnuradio/gr-digital/lib/CMakeLists.txt
pybombs/bladeRF/src/gnuradio/gr-digital/python/digital/bindings/CMakeLists.txt 
pybombs/bladeRF/src/gnuradio/gr-dtv/lib/CMakeLists.txt
pybombs/bladeRF/src/gnuradio/gr-channels/lib/CMakeLists.txt
pybombs/bladeRF/src/gnuradio/gr-qtgui/lib/CMakeLists.txt

Regards,
Marco

CeSiumUA, 2024-12-11:
Hi, @marco-bv !
Thanks a lot for that workaround. I haven't had a chance to try it yet, as I've already installed bladeRF cli via a repository, but I'll definitely try to build it with PyBOMBS with your fix.

marco-bv, 2024-12-12:
Hello @CeSiumUA,

I completed the ubuntu 24.04 installation. If you have some issue while trying to build lmk.

Regards

rghilduta, 2025-05-20:
Instructions have been updated! Issue should be resolved, please reopen/comment if it has not

closed 2025-05-20 by rghilduta