Nuand/bladeRF #992: Unable to load FPGA image

andersonreloaded, 2024-11-28:
Hi Folks,

I am relatively new to the ecosystem of bladeRF SDR setup. Currently using the latest git version of bladeRF on Ubuntu 24.04 LTS. Have been trying to load FPGA image for bladerf 2.0 A4 using the command 'bladerf-cli' with following error.

> 
> [VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:642] Using libusb version: 1.0.27.11882
> [VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:519] Found a bladeRF (idx=1)
> [DEBUG @ host/libraries/libbladeRF/src/backend/usb/libusb.c:189] Bus 001 Device 005: Nuand bladeRF 2.0, serial xxx
> [VERBOSE @ host/libraries/libbladeRF/src/backend/usb/libusb.c:581] USB port reset succeeded for bladeRF xxx
> [VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 0
> [VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:293] Read Firmware version: 2.4.0-git-a3d5c55f
> [VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:299] Capability mask before FPGA load: 0x0000007f00000000
> [VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:401] Found SPI flash manufacturer: WINBOND.
> [VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:404] Found SPI flash device: W25Q32JV (32 Mbit).
> [VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:452] SPI flash total size = 32 Mbit
> [VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:453] SPI flash page size = 256 bytes
> [VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:454] SPI flash erase block size = 65536 bytes
> [VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:455] SPI flash number of pages = 16384
> [VERBOSE @ host/libraries/libbladeRF/src/board/bladerf1/flash.c:456] SPI flash number of erase blocks = 64 pages
> [VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 2
> [VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 3
> [DEBUG @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:447] Loading FPGA from: /usr/share/Nuand/bladeRF/hostedxA4.rbf
> [VERBOSE @ host/libraries/libbladeRF/src/backend/usb/usb.c:106] Changing to USB alt setting 3
> [DEBUG @ host/libraries/libbladeRF/src/backend/usb/libusb.c:950] lusb_control_transfer failed: status = -7
> [DEBUG @ host/libraries/libbladeRF/src/backend/usb/usb.c:430] Failed to initiate FPGA programming: Operation timed out
> [ERROR @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:457] bladerf2_open: dev->backend->load_fpga(dev, buf, buf_size) failed: Operation timed out
> [VERBOSE @ host/libraries/libbladeRF/src/device_calibration.c:248] Freeing gain calibration table
> [VERBOSE @ host/libraries/libbladeRF/src/device_calibration.c:248] Freeing gain calibration table
> [VERBOSE @ host/libraries/libbladeRF/src/device_calibration.c:248] Freeing gain calibration table
> [VERBOSE @ host/libraries/libbladeRF/src/device_calibration.c:248] Freeing gain calibration table
> Failed to open device (first available): Operation timed out 
> 

Any suggestion or assistance is highly appreciated.

C-STACK7, 2026-11-12:
Hello!!!
This problem is related to the VCTCXO calibration data.
It seems that the data has been erased from the flash, and the checksum doesn’t match. You need to download the source code, add this text to the bladeRF-cli in the bladerf2.c file. Rebuild the source code. Run the assembled part with the  firmware bladeRF-cli -L <*.rbf>. Then the FPGA will be flashed. After that, find out the calibration data on the Nuand website. Enter the new calibration data. Rebuild the project in its old form. This will bring the device to life. Good luck.

/* Get FPGA size */
    status = spi_flash_read_fpga_size(dev, &board_data->fpga_size);
    if (status < 0) {
        log_warning("Failed to get FPGA size %s\n", bladerf_strerror(status));
    }

    if (getenv("BLADERF_FORCE_FPGA_A9")) {
        log_info("BLADERF_FORCE_FPGA_A9 is set, assuming A9 FPGA\n");
        board_data->fpga_size = BLADERF_FPGA_A9;
    }
else {
                log_info("BLADERF_FORCE_FPGA_A4 is set, assuming A4 FPGA\n");
        board_data->fpga_size = BLADERF_FPGA_A4;
}

[image]

C-STACK7, 2026-11-12:
This is a temporary workaround for bladeRF-cli; after recovery, it needs to be returned to the standard version of bladeRF-cli. If you can’t do it, I can send you the workaround file.