Nuand/bladeRF #994: build fails with gcc-14

paparodeo, 2024-12-06:
> host/utilities/bladeRF-fsk/c/src/fir_filter.c:227:28:
error: 'calloc' sizes specified with 'sizeof' in the earlier argument and not in the later argument 
  227 |     outbuf = calloc(sizeof(struct complex_sample), chunk_size);

https://cache.nixos.org/log/hsg1v3cl1hk9ca77rgkng6n6dabz1bi3-libbladeRF-2.5.0.drv

transporting the args to `calloc` so `chunk_size` is passed as the number of elements (which seems correct given the call to tread below).

```diff
diff --git a/host/utilities/bladeRF-fsk/c/src/fir_filter.c b/host/utilities/bladeRF-fsk/c/src/fir_filter.c
index 59f34f0..7def697 100644
--- a/host/utilities/bladeRF-fsk/c/src/fir_filter.c
+++ b/host/utilities/bladeRF-fsk/c/src/fir_filter.c
@@ -213,18 +213,18 @@ int main(int argc, char *argv[])
         return EXIT_FAILURE;
     }
 
-    inbuf = calloc(2*sizeof(int16_t), chunk_size);
+    inbuf = calloc(chunk_size, 2*sizeof(int16_t));
     if (!inbuf) {
         perror("calloc");
         goto out;
     }
-    tempbuf = calloc(2*sizeof(int16_t), chunk_size);
+    tempbuf = calloc(chunk_size, 2*sizeof(int16_t));
     if (!tempbuf) {
         perror("calloc");
         goto out;
     }
 
-    outbuf = calloc(sizeof(struct complex_sample), chunk_size);
+    outbuf = calloc(chunk_size, sizeof(struct complex_sample));
     if (!outbuf) {
         perror("calloc");
         goto out;
```

paparodeo, 2024-12-06:
duplicate of https://github.com/Nuand/bladeRF/issues/972
PR for fix is https://github.com/Nuand/bladeRF/pull/984

closed 2025-01-12 by paparodeo

FoxRomeo, 2025-01-26:
Hi
is there a planned date for a merge of this PR or anything that prevents the merge?