Nuand/bladeRF #995: macOS 15.2 - Could NOT find Curses (missing: CURSES_INCLUDE_PATH)

kaipyroami, 2024-12-23:
Building from source using homebrew yields this error on the `cmake -DCMAKE_BUILD_TYPE=Debug ../` step.
I have been unable to find where CURSES_INCLUDE_PATH is defined or where it should be defined.

`brew --prefix ncurses` provides the following directory for curses, 
`/opt/homebrew/opt/ncurses`

Error:
```
CMake Error at /opt/homebrew/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:233 (message):
  Could NOT find Curses (missing: CURSES_INCLUDE_PATH)
Call Stack (most recent call first):
  /opt/homebrew/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:603 (_FPHSA_FAILURE_MESSAGE)
  /opt/homebrew/share/cmake/Modules/FindCurses.cmake:268 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  utilities/bladeRF-power/CMakeLists.txt:6 (find_package)
```

kaipyroami, 2024-12-23:
Attempted to set the path in `utilities/bladeRF-power/CMakeLists.txt`
with 
`find_package(Curses REQUIRED PATHS "/opt/homebrew/opt/ncurses")`

Error:
```
CMake Error at utilities/bladeRF-power/CMakeLists.txt:6 (find_package):
  Could not find a package configuration file provided by "Curses" with any
  of the following names:

    CursesConfig.cmake
    curses-config.cmake

  Add the installation prefix of "Curses" to CMAKE_PREFIX_PATH or set
  "Curses_DIR" to a directory containing one of the above files.  If "Curses"
  provides a separate development package or SDK, be sure it has been
  installed.
```

ben-morse-plastic, 2025-06-17:
Please try again on master (https://github.com/Nuand/bladeRF/commit/4494aa41abbc8d24d4b49dce7eb8eb474c123fe7)

rthomp10, 2026-10-17:
Fixed in b2f9503fc

closed 2026-10-17 by rthomp10