Nuand/bladeRF #1000: bladerf_schedule_retune behaves incorrectly!

GvozdevLeonid, 2025-02-07:
For some reason I can only switch really fast between 9 frequencies. If I add 10 frequencies and more, the switching time becomes very long and I can't accurately predict the time to get samples.

Here are the logs for 9 frequency switches:
```
 create tune:70.0 real:100.0 device:100.0
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1450] Quick tune assigned Nios RX fast lock index: 0
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1454] Quick tune assigned RFFE RX fast lock index: 0
 create tune:130.0 real:160.0 device:160.0
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1450] Quick tune assigned Nios RX fast lock index: 1
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1454] Quick tune assigned RFFE RX fast lock index: 1
 create tune:190.0 real:220.0 device:219.999998
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1450] Quick tune assigned Nios RX fast lock index: 2
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1454] Quick tune assigned RFFE RX fast lock index: 2
 create tune:250.0 real:280.0 device:279.999998
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1450] Quick tune assigned Nios RX fast lock index: 3
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1454] Quick tune assigned RFFE RX fast lock index: 3
 create tune:310.0 real:340.0 device:339.999998
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1450] Quick tune assigned Nios RX fast lock index: 4
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1454] Quick tune assigned RFFE RX fast lock index: 4
 create tune:370.0 real:400.0 device:400.0
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1450] Quick tune assigned Nios RX fast lock index: 5
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1454] Quick tune assigned RFFE RX fast lock index: 5
 create tune:430.0 real:460.0 device:459.999998
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1450] Quick tune assigned Nios RX fast lock index: 6
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1454] Quick tune assigned RFFE RX fast lock index: 6
 create tune:490.0 real:520.0 device:520.0
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1450] Quick tune assigned Nios RX fast lock index: 7
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1454] Quick tune assigned RFFE RX fast lock index: 7
 create tune:550.0 real:580.0 device:579.999998
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1450] Quick tune assigned Nios RX fast lock index: 8
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/bladerf2.c:1454] Quick tune assigned RFFE RX fast lock index: 0
 Sweeping from 70.0 MHz to 610.0 MHz
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:429] nios_config_read: Read 0x00010001
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:440] nios_config_write: Wrote 0x00010001
[VERBOSE @ host/libraries/libbladeRF/src/board/bladerf2/common.c:353] check_total_sample_rate: active_channels=1, rate_accum=60000000, maximum=80000000
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:502] nios_get_timestamp: Read RX timestamp: 846905
 setup tune:70.0 time_start:9846905
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1180] nios_retune2: channel=RX1 timestamp=9846905 nios_profile=0 rffe_profile=0
                                port=0x8c spdt=0x0a
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1198] RX1 operation duration: 58852
 setup tune:130.0 time_start:11048021
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1180] nios_retune2: channel=RX1 timestamp=11048021 nios_profile=1 rffe_profile=1
                                port=0x8c spdt=0x0a
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1198] RX1 operation duration: 1090
 setup tune:190.0 time_start:12249137
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1180] nios_retune2: channel=RX1 timestamp=12249137 nios_profile=2 rffe_profile=2
                                port=0x8c spdt=0x0a
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1198] RX1 operation duration: 1452
 setup tune:250.0 time_start:13450253
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1180] nios_retune2: channel=RX1 timestamp=13450253 nios_profile=3 rffe_profile=3
                                port=0x8c spdt=0x0a
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1198] RX1 operation duration: 1817
 setup tune:310.0 time_start:14651369
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1180] nios_retune2: channel=RX1 timestamp=14651369 nios_profile=4 rffe_profile=4
                                port=0x8c spdt=0x0a
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1198] RX1 operation duration: 2182
 setup tune:370.0 time_start:15852485
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1180] nios_retune2: channel=RX1 timestamp=15852485 nios_profile=5 rffe_profile=5
                                port=0x8c spdt=0x0a
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1198] RX1 operation duration: 2549
 setup tune:430.0 time_start:17053601
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1180] nios_retune2: channel=RX1 timestamp=17053601 nios_profile=6 rffe_profile=6
                                port=0x8c spdt=0x0a
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1198] RX1 operation duration: 2916
 setup tune:490.0 time_start:18254717
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1180] nios_retune2: channel=RX1 timestamp=18254717 nios_profile=7 rffe_profile=7
                                port=0x8c spdt=0x0a
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1198] RX1 operation duration: 3287
 setup tune:550.0 time_start:19455833
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1180] nios_retune2: channel=RX1 timestamp=19455833 nios_profile=8 rffe_profile=0
[... 625 lines cut ...]
 setup tune:190.0 time_start:107705088
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1180] nios_retune2: channel=RX1 timestamp=107705088 nios_profile=2 rffe_profile=2
                                port=0x8c spdt=0x0a
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1198] RX1 operation duration: 3646
 get samples:190.0 time_at:98695072
 setup tune:250.0 time_start:108906340
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1180] nios_retune2: channel=RX1 timestamp=108906340 nios_profile=3 rffe_profile=3
                                port=0x8c spdt=0x0a
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1198] RX1 operation duration: 3646
 get samples:250.0 time_at:99896324
 setup tune:310.0 time_start:110107592
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1180] nios_retune2: channel=RX1 timestamp=110107592 nios_profile=4 rffe_profile=4
                                port=0x8c spdt=0x0a
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1198] RX1 operation duration: 3646
 get samples:310.0 time_at:101097576
 setup tune:370.0 time_start:111308844
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1180] nios_retune2: channel=RX1 timestamp=111308844 nios_profile=5 rffe_profile=5
                                port=0x8c spdt=0x0a
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1198] RX1 operation duration: 3646
 get samples:370.0 time_at:102298828
 setup tune:430.0 time_start:112510096
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1180] nios_retune2: channel=RX1 timestamp=112510096 nios_profile=6 rffe_profile=6
                                port=0x8c spdt=0x0a
[VERBOSE @ host/libraries/libbladeRF/src/backend/usb/nios_access.c:1198] RX1 operation duration: 3646
```

rthomp10, 2025-02-11:
The first 8 fastlock profiles are preloaded onto the AD9361. Subsequent profiles are stored in Nios memory. After the first 8 profiles are iterated through, the Nios core hot-loads the ones in Nios memory to the AD9361 which takes just under millisecond per profile.

GvozdevLeonid, 2025-02-12:
@rthomp10 
May you tell me how loading from Nios works? As I understand, I can't set `bladerf_schedule_retune` for different quick_tune and different timestamp but with the same profile index, right? It seems like the new bladerf_schedule_retune overwrites the quick tuning profile in AD9361.

Also, according to my tests, switching takes from 5 to 6 milliseconds (my delay is calculated as 
`await_time = (sample_rate // 1000) * 6`
For some reason, even if I call bladerf_schedule_retune much earlier, I still have to wait for these 6 milliseconds, although I expected that the switching time would be lower, since it seems that the data from quick_tune is written to the AD9361 immediately after calling `bladerf_schedule_retune`.

I implemented the sweep mode and noticed that with a delay of less than 6 milliseconds, I can get samples from the old frequency range.

GvozdevLeonid, 2025-05-26:
Hello. I recently noticed another bug that is a bit confusing. If I create 9 quicktune profiles, the 9th one will be the profile to replace the first profile in fastlock. But if I do something like this:

bladerf_schedule_retune(channel, schedule_timestamp, start_from + offset, quick_tune_8)
and then immediately
bladerf_schedule_retune(channel, schedule_timestamp, start_from + offset, quick_tune_0)

Then, since quick_tune_0 and quick_tune_8 contain the same address in fastlock, the last bladerf_schedule_retune will simply overwrite the previous frequency.

I studied the source code a bit and noticed that there is a duration variable that probably contains the required time to wait for the next frequency, but I can't access it, and I would like to have this function so that there is a minimum delay

GvozdevLeonid, 2025-05-26:
I found a simple way to get around this (but I lose 3 quick_tune profiles). First you need to calculate the required number of quick_tune profiles.

My code is in python, but it will be easy to rework for other languages.
```py
frequencies: list[float] = [your frequencies]
quick_tune_profiles = []

shift_after = -1
if len(frequencies) % 8 < 4:
    shift_after = len(frequencies) - len(frequencies) % 8

for i, frequency in enumerate(frequencies):
    if i == shift_after:
        device.pybladerf_get_quick_tune(channel)
        device.pybladerf_get_quick_tune(channel)
        device.pybladerf_get_quick_tune(channel)

    device.pybladerf_set_frequency(channel, frequency + offset)
    quick_tune_profiles.append(device.pybladerf_get_quick_tune(channel))
```

then your retune queue  must contain 4 or less profiles at a time