Nuand/bladeRF #1001: masos sequoia 15.3.1 macports & build from source installation fails

oleksiipysanko, 2025-02-16:
Tecla build fails in macports:

```
:info:build make[1]: Entering directory `/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_devel_tecla/tecla/work/libtecla'
:info:build /usr/bin/clang -c -O  -DPACKAGE_NAME="" -DPACKAGE_TARNAME="" -DPACKAGE_VERSION="" -DPACKAGE_STRING="" -DPACKAGE_BUGREPORT="" -DPACKAGE_URL="" -DSTDC_HEADERS=1 -DHAVE_SYS_TYPES_H=1 -DHAVE_SYS_STAT_H=1 -DHAVE_STDLIB_H=1 -DHAVE_STRING_H=1 -DHAVE_MEMORY_H=1 -DHAVE_STRINGS_H=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_UNISTD_H=1 -DUSE_TERMINFO=1 -DHAVE_CURSES_H=1 -DHAVE_TERM_H=1 -DHAVE_SYS_SELECT_H=1 -DHAVE_SELECT=1 -DHAVE_SYSV_PTY=1 -pipe -Os -isysroot/Library/Developer/CommandLineTools/SDKs/MacOSX15.sdk -arch arm64  -o normal_obj/getline.o ./getline.c
:info:build ./getline.c:3931:34: error: incompatible function pointer types passing 'TputsRetType (TputsArgType)' (aka 'void (int)') to parameter of type 'int (*)(int)' [-Wincompatible-function-pointer-types]
:info:build  3931 |     tputs((char *)string, nline, gl_tputs_putchar);
:info:build       |                                  ^~~~~~~~~~~~~~~~
:info:build /opt/local/include/term.h:848:60: note: passing argument to parameter here
:info:build   848 | extern NCURSES_EXPORT(int) tputs (const char *, int, int (*)(int));
:info:build       |                                                            ^
:info:build 1 error generated.
:info:build make[1]: *** [normal_obj/getline.o] Error 1
:info:build make[1]: Leaving directory `/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_devel_tecla/tecla/work/libtecla'
:info:build make: *** [normal] Error 2
:info:build make: Leaving directory `/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_devel_tecla/tecla/work/libtecla'
:info:build Command failed:  cd "/opt/local/var/macports/build/_opt_local_var_macports_sources_rsync.macports.org_macports_release_tarballs_ports_devel_tecla/tecla/work/libtecla" && /usr/bin/make -w default 
:info:build Exit code: 2
```

Build from source fails in bladerf-power:

```
[ 96%] Linking C executable ../../output/bladeRF-power
Undefined symbols for architecture arm64:
  "_doupdate$NCURSES60", referenced from:
      _update_window in window.c.o
      _display_overlay in window.c.o
  "_endwin$NCURSES60", referenced from:
      _start_streaming in helpers.c.o
  "_initscr$NCURSES60", referenced from:
      _init_curses in window.c.o
  "_keypad$NCURSES60", referenced from:
      _init_curses in window.c.o
  "_mvwprintw$NCURSES60", referenced from:
      _update_window in window.c.o
      _update_window in window.c.o
      _update_window in window.c.o
      _update_window in window.c.o
      _update_window in window.c.o
      _update_window in window.c.o
      _update_window in window.c.o
      _update_window in window.c.o
      _update_window in window.c.o
      _update_window in window.c.o
      _update_window in window.c.o
      _update_window in window.c.o
      _update_window in window.c.o
      ...
  "_newwin$NCURSES60", referenced from:
      _init_curses in window.c.o
      _display_overlay in window.c.o
  "_waddnstr$NCURSES60", referenced from:
      _display_digit in text.c.o
      _display_digit in text.c.o
      _display_digit in text.c.o
      _display_digit in text.c.o
      _display_digit in text.c.o
      _display_digit in text.c.o
      _display_double in text.c.o
      _display_double in text.c.o
      _display_double in text.c.o
      _display_double in text.c.o
      _display_double in text.c.o
      _display_double in text.c.o
      ...
  "_wborder$NCURSES60", referenced from:
      _update_window in window.c.o
      _display_overlay in window.c.o
  "_wgetch$NCURSES60", referenced from:
      _start_streaming in helpers.c.o
  "_wnoutrefresh$NCURSES60", referenced from:
      _update_window in window.c.o
      _display_overlay in window.c.o
ld: symbol(s) not found for architecture arm64
```

ArhumJain, 2025-05-16:
Experienced the same error, fixed it by changing line 3931 from 

`tputs((char *)string, nline, gl_tputs_putchar);`

to

`tputs((char *)string, nline, (int*)  &gl_tputs_putchar);`

ArhumJain, 2025-05-16:
I made a pull request but as the repo hasn't been updated in years, may just have to deal with the manual fix:

https://github.com/ianliu/libtecla/commit/33b248342890a22fa5ad6038268d156e9151c5c6

rghilduta, 2025-05-16:
hmm bladeRF-cli should probably just move to readline

rghilduta, 2025-05-16:
Actually.... libedit seems fairly well maintained.

I'd suggest trying libedit versus libtecla by running:

sudo apt-get install libedit-dev
(from host/build) cmake .. -DENABLE_LIBEDIT=ON

Now I thinking maybe make libedit the default