Nuand/bladeRF #1006: 403: Forbidden from GitHub actions

tarasktr, 2025-04-07:
I run CI on GitHub runners, and when accessing nuand.com get 403 error

```
wget https://www.nuand.com/fx3/bladeRF_fw_v2.4.0.img

Resolving www.nuand.com (www.nuand.com)... 104.26.13.238, 172.67.69.95, 104.26.12.238, ...
Connecting to www.nuand.com (www.nuand.com)|104.26.13.238|:443... connected.
HTTP request sent, awaiting response... 403 Forbidden
2025-04-07 10:44:02 ERROR 403: Forbidden.
```

Are there some rate limits for GitHub IPs or something else? How can I install the bladeRF package from the GitHub Actions Job?

rghilduta, 2025-04-18:
This should have resolved itself. We are behind Cloudflare so it might have caught a few extra things it shouldn't have for a few days.

closed 2025-04-18 by rghilduta

tarasktr, 2025-04-18:
I can confirm it works now. Thank you