Nuand/bladeRF #1015: App crashes on first synchronous RX after clean install on Windows 11

vadimart92, 2025-05-14:
The application crashes on the first run when attempting to receive data using the synchronous API on Windows 11 after a clean install. However, the second run works without any issues.

Preconditions:

Clean install of Windows 11.

Connect a bladeRF2 device and install [driver](https://github.com/Nuand/bladeRF/releases/tag/2024.05)

Run app that uses the synchronous API for RX:
bladerf_sync_config
bladerf_enable_module
bladerf_sync_rx

Observed Behavior:
bladerf_sync_rx times out.
The process terminates due to an Access Violation in set_state, which is called from sync_worker_task.
[image]
which is called from sync_worker_task
[image]

But second time you run same app - it may just work (not always).

The crash may be related to a missing MSVCR100.dll dependency. The pthreadVC2.dll depends on MSVCR100.dll, which is not included in the Visual C++ Redistributable for Visual Studio 2015–2022.

rghilduta, 2025-05-14:
@vadimart92 we actually just got rid of the pthreadsVC2.dll dependency in libbladeRF! please give us a few more days to generate a Windows release candidate. If you do not see one by Friday please tag me on this thread.

vadimart92, 2025-05-14:
Thats fast ;)

vadimart92, 2025-05-16:
@rghilduta i've checked on tag 2025.05-rc1 - still same problem, probably worker thread gets this mutex lock later than state de-initialised, seems like simple null check could fix this issue

rghilduta, 2025-05-20:
Try this: https://github.com/Nuand/bladeRF/releases/tag/2025.05-rc2

vadimart92, 2025-05-21:
Issue persists.
On a VM with bladeRF-cli 2.5.0, I ran installer 2025.05-rc2 (with connected device) and rebooted. 
An interesting observation: the issue reproduces only if I run the app immediately after reboot (while the system is under some load). However, if I wait for about 10 minutes, the app works fine.