Nuand/bladeRF #1017: Need to add libcurl-devel as dependency for Redhat-based distros (e.g., Fedora, CentOS) in the wiki

fahadwajed, 2025-05-19:
Need to add _libcurl-devel_ as dependency in **[Redhat-based distros (e.g., Fedora, CentOS)](https://github.com/Nuand/bladeRF/wiki/Getting-Started%3A-Linux#user-content-Installing_dependencies)** otherwise new user might find them seeing this error
```
CMake Error at /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:233 (message):
  Could NOT find CURL (missing: CURL_LIBRARY CURL_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:603 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake/Modules/FindCURL.cmake:203 (find_package_handle_standard_args)
```

Also _dnf_ is the default package manager for Fedora since Fedora 22. Need to update that as well.

rghilduta, 2025-05-19:
will add!

rghilduta, 2025-05-20:
added and fixed!

closed 2025-05-20 by rghilduta