Nuand/bladeRF #1019: Build error libbladeRF_test_repeater

ritabarr, 2025-05-24:
Error while trying to build following the instructions at https://github.com/Nuand/bladeRF/wiki/Getting-Started%3A-Linux#user-content-Configure_the_build

Please advise on how to correct or bypass. 

Running kali linux:
    PRETTY_NAME="Kali GNU/Linux Rolling"
    NAME="Kali GNU/Linux"
    VERSION_ID="2025.2"
    VERSION="2025.2"
    VERSION_CODENAME=kali-rolling
    ID=kali
    ID_LIKE=debian
    HOME_URL="https://www.kali.org/"
    SUPPORT_URL="https://forums.kali.org/"
    BUG_REPORT_URL="https://bugs.kali.org/"
    ANSI_COLOR="1;31"

=========================================

**Step: Perform the build and installation**
The following commands:

1. Perform the build
2. Install files to /usr/local or the location specified by CMAKE_INSTALL_PREFIX
3. Updated share library paths, so that libbladeRF can be found

         $ make && sudo make install && sudo ldconfig

======================================================
Fails on the following **_out of bounds error_** which is probably triggering the 2nd error: 

[ 41%] Building C object libraries/libbladeRF_test/test_repeater/CMakeFiles/libbladeRF_test_repeater.dir/src/repeater.c.o
In function ‘term_init’,
    inlined from ‘repeater_run’ at /root/bladeRF/host/libraries/libbladeRF_test/test_repeater/src/repeater.c:790:14:
/root/bladeRF/host/libraries/libbladeRF_test/test_repeater/src/repeater.c:115:5: **error: array subscript ‘unsigned char[60][0]’ is partly outside array bounds of ‘struct termios[0]’ [-Werror=array-bounds=]
  115 |     memcpy(&termios_new, &termios_backup, sizeof(termios_new));
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~**
/root/bladeRF/host/libraries/libbladeRF_test/test_repeater/src/repeater.c: In function ‘repeater_run’:
/root/bladeRF/host/libraries/libbladeRF_test/test_repeater/src/repeater.c:96:24: note: object ‘termios_backup’ of size 8
   96 | static struct termios *termios_backup = NULL;
      |                        ^~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [libraries/libbladeRF_test/test_repeater/CMakeFiles/libbladeRF_test_repeater.dir/build.make:93: libraries/libbladeRF_test/test_repeater/CMakeFiles/libbladeRF_test_repeater.dir/src/repeater.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1312: libraries/libbladeRF_test/test_repeater/CMakeFiles/libbladeRF_test_repeater.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

rghilduta, 2025-06-08:
please ensure you are on master HEAD, there have been a number of fixes including 0ffb795c450fe814060f95cd37455847c9c536d2 which addresses this memcpy()

rthomp10, 2026-08-28:
Fixed since commit [0ffb795c](https://github.com/Nuand/bladeRF/commit/0ffb795c).

closed 2026-08-28 by rthomp10